atanisoft/ESP32CommandStation #1: document building with Arduino IDE

atanisoft, 2017-12-11:
Currently the code is only built and tested using the Platform IO IDE. Instructions will need to be provided for using Arduino IDE as an option for users.

mijnmodelbaan, 2018-02-01:
Using the Arduino IDE would be great, but it is a lot of work, I think.

atanisoft, 2018-02-02:
Yes, there are a number of steps and some external tools are required due to the html being embedded in the code.

mijnmodelbaan, 2018-02-02:
I started compiling the code in the Arduino IDE but I'm missing the "index_html.h" file; it's not in the repository. Can you add it, please, so I can plow on? Thanks!

atanisoft, 2018-02-02:
You can generate that file manually via these steps:
`gzip -c data/index.html > index_html.gz
bin2c index_html.gz index_html.h indexHtmlGz`

You will need to open the index_html.h file generated from above and make a couple small adjustments to it:
`const int indexHtmlGz_length =`
will need to be changed to:
`const size_t indexHtmlGz_size =`

and
`const char indexHtmlGz[???] =`
should be changed to:
`const uint8_t indexHtmlGz[] PROGMEM =`
Note the removal of the number inside the [] marks and addition of PROGMEM.

If you don't have bin2c you can grab it from https://github.com/gwilymk/bin2c or you can look at build_index_header.py in the DCCppESP32 source tree for how this is done with PlatformIO IDE automatically.

atanisoft, 2019-02-17:
Using the Arduino IDE is not really an option unfortunately as of the v1.2.0 release. The Arduino IDE does not support a custom partition scheme (larger app, OTA and spiffs) nor does it support the index_html.h generation.

It is highly recommended to use another IDE that is more full featured.

closed 2019-02-17 by atanisoft