atanisoft/ESP32CommandStation #2: Implement S88 Sensor support

atanisoft, 2017-12-11:
This will provide up to 496 sensors with only a few GPIO pins required to read state of the inputs.

atanisoft, 2017-12-12:
http://www.s88-n.eu/s88-timing-en.html

atanisoft, 2017-12-12:
http://www.s88-n.eu/index-en.html

zubozrout, 2019-10-19:
Hi,
sorry to ask under this issue, but I was trying to find out more about the wiring needed for that and haven't been able to find a single resource for that.

All I have found is this page:
https://atanisoft.github.io/ESP32CommandStation/config-s88.html
plus that I will most certainly need a RJ45 connector to easily connect all S88/S88-N sensors.

Unfortunately I haven't found a single wiring scheme. Could you please share more details on that, perhaps add it to the documentation page?

Info like: what components should one buy, where should it get connected to on the ESP32 board and maybe some simple scheme would help a lot.

Thank you very much.

atanisoft, 2019-10-19:
@zubozrout I've been meaning to add more documentation all around on the site. I'll add some details on the wiring. For S88 you would likely be best with an RJ45 breakout board for the connections to the esp32. Three pins are common across all S88 buses with one additional pi per bus for data. I've got a reference page for the S88 pinout for RJ45 and the inline pin connector and will add both to the webpage today.

atanisoft, 2019-10-19:
@zubozrout I've updated the page to now include more details. It may take a short bit for it to replicate to the webservers.

zubozrout, 2019-10-19:
Thank you very much for all the info and the updated wiki, really appreciate it :).

zubozrout, 2020-02-08:
Hello,
I finally got around to wiring everything up (properly). I hope I got everything right - in regards to the default configuration at least. I've also uncommented `#include "Config_S88.h"` in Config.h. But I am unsure about the following:

`NOTE: The S88 Bus will require a data pin in addition to the pins that are common to all S88 Buses.` which I am sure I haven't done.

It might be related to this `<S88 ID DATAPIN COUNT> : Creates an S88 Sensor Bus with the specified ID, DATA PIN, SENSOR COUNT.` but how should I trigger that S88 command (where)?

All I have available is the DCCppESP32 Web interface and JMRI. And then a occupancy detector connected to the S88N bus.

Also, once the data pin is configured, what should I see? Will there be a new item for S88 sensors in the Web UI and/or where to look for that in JRMI?

Thank you very much and sorry to be bothering you with these for sure trivial things to solve.

atanisoft, 2020-02-08:
If you enable the S88 support in the code, recompile and flash it to the esp32 you should see a S88 section available on the configuration tab of the web interface.

From the web interface you should be able to create the S88 Bus with the data pin and how many S88 inputs are expected. Sending the S88 command as written above is not directly supported from JMRI.

Once the S88 Bus has been defined, the inputs will show up as regular sensors in JMRI with a base ID of 512 (default) and count upwards from there for each input. Secondary buses will do the same but will start with the first sensor id as a multiple of 512.

commit dda34884 (2019-06-04): RC data fix #2....