atanisoft/ESP32CommandStation #4: AP creation when not at 'home' network

mijnmodelbaan, 2018-02-06:
**Request** 
to let the ESP create her own Access Point when it can't connect to the in config.h defined network after, say, 5 or 10 tries.  This seems to me very useful if you want to take the layout to a show/exhibition. The SSID and password should be configurable.

atanisoft, 2018-02-06:
This is a great idea. I have been thinking on how to move away from needing Config.h modifications to reconfigure the base station features. I will look into what is needed for this so we can setup a captive portal on the AP with the option of "Soft AP" as a default option as well.

atanisoft, 2018-02-09:
@mijnmodelbaan I worked on an implementation of this and it half worked, the arduino-esp32 libraries are a bit unstable at the moment for DNSServer so this will need to be revisited after that has been fixed. I also ran into the case where the web interface references multiple external sites and those are not available when in SoftAP mode so an SD card will be required for the web interface unless it gets a redesign (which it is needing a bit of anyway).  So for now this will be pushed out to a later release.

closed 2019-10-20 by atanisoft