atanisoft/ESP32CommandStation #5: Integration of LocoNet

mijnmodelbaan, 2018-02-06:
**Request**
to integrate LocoNet in the system. There are already several LocoNet shields (Arduino footprint) available and also the necessary sketches for operating points and accessories.

atanisoft, 2018-02-06:
LocoNet is on the roadmap but I don't have any LN hardware that I can test with currently.  If you are willing/able to help test it I can start implementing this sooner.

atanisoft, 2018-02-06:
One issue with the LocoNet libraries that are available is that they are currently only supporting AVR architectures. There is one ESP32 compatible library but it is not usable since it would prevent the DCC signal from being generated due to overlap on usage of the hardware timers. An alternative library will need to be developed and that will take time unfortunately.

afaber999, 2018-03-02:
Hi great work so far, I was planning to port DCC++ to ESP32 as well and also add support for Loconet.
There is an effort going to port the loconet library to ESP32 (more information on the loconet_hackers yahoo group). An inital port can be found at:

https://github.com/mcmadhatter/LocoNet2/tree/development

atanisoft, 2018-03-02:
@afaber999, unfortunately that is not usable as it uses the hardware timers that are used to generate the dcc signal. So a version of this for the esp32 will be needed that does not depend on the hardware timers.

I am not sure what option will work for this just yet. But one is switch from two timers to just one for main and prog (currently it is one for full wave and one for half wave).

afaber999, 2018-03-02:
OK, I guess some research is required , some ideas
      Maybe it is possible to use the I2S hardware to generate the DCC signals
      Make use the LED PWM timers
      Maybe it is possible to use the UART for Loconet
      .....

atanisoft, 2018-03-02:
I started with ledc and it didn't work correctly for the signal. Not sure on i2s but with timers it works perfectly.

I am interested in using the second uart for LocoNet but also looking at using it for a nextion display as an integrated console. More research is needed still. Switching to using only two timers vs four is likely the easiest but will need a bit of testing to confirm the signal is correct. If I can get that working the LocoNet can be used as long as the timer is configurable in the library (or modify to have it locally implemented)

afaber999, 2018-03-07:
Hi, can you indicate why the LEDC did not work out? I think it is feasible to use the LEDC_HSTIMER0_OVF_INT interrupt to change the period length to toggle between the logic levels

atanisoft, 2018-03-07:
I couldn't get a stable signal for both prog and main with ledc. It may be possible but it wasn't something that I could make work reliably with the arduino-esp32 code.

There are other options for LocoNet that I am looking into that may be usable but they will be a bit down the road. I don't have any LocoNet hardware to test with so it will be very slow going.

kiwi64ajs, 2018-06-10:
Hi Guys, Alex Shepherd here (Author of MRRwA Arduino NmraDcc and LocoNet libraries) and I was just wondering about using an ESP32 for DCC++ and just found this project...

While studying the ESP32 hardware recently I found it has a built-in "Remote Peripheral" that is intended for Sending/Receiving Infrared Remote signals and has onboard 38kHz modulation etc. It can be configured to support 8 Channels that get their bit timings from a shared array of 512 x 32-bit timer values that can be configured to generate plain or modulated output signals. It occurs to me that this would be ideal for generating a DCC packet. Using the Remote Peripheral for DCC Signal generation should relieve the ESP32 CPU usage significantly. 

As stated already, it would also be good to support LocoNet and others have already begun to port my LocoNet code to LocoNet2 to de-AVR-ize it which would be really good.

I've also done a bit of work to write a WiThrottle Server using the ESP8266 with the thought of making a WiThrottle to LocoNet gateway device but not gotten very far with it...

Anyway thanks for your work and let me know if you need help testing on LocoNet hardware

Regards

Alex Shepherd

atanisoft, 2018-06-10:
@kiwi64ajs I'll check into that. I am not sure that Remote Peripheral has been exposed in the arduino-esp32 code. The timers approach is working well for now and doesn't tax the CPU very heavily. The biggest problem is the IRQ needed to change the width/direction based on the DCC bits.

For LocoNet, if it didn't depend on a timer (or IRQ for that matter) it would likely be more easily integrated. The esp32 has a real problem with the IRQ firing on one CPU and the other accessing flash memory (maybe even sd card, haven't tested yet) and causing a CPU crash. So if it needs a timer we would need to be able to control that so it can be stopped when accessing flash memory.

atanisoft, 2018-10-10:
@kiwi64ajs revisiting this again as I have seen some code that uses the RMT module to generate a DCC signal. I'm looking into how that can be utilized now so the LocoNet2 library may end up as a viable option but I don't have any way to test the LN support. If I can get the RMT module to generate a signal reliably that will free up the timers and make it very feasible to use a timer for the LN code.

atanisoft, 2019-02-13:
@mijnmodelbaan experimental support is available on the development branch, it is not fully tested as I lack the hardware to test it fully at this time. I'm going to include it in the v1.2.0 release that is planned for this coming weekend and will keep this issue open until I have tested it fully and implemented the remaining integration points (switch req/rep, input rep, prog track).

alex-shepherd-nv, 2019-04-22:
Another LocoNet library issue is that for a Command Station there are some subtle differences in behaviour, mostly relating to the CD_Backoff AND the need to send OPC_BUSY packets if the CD_Backup expires while the Command Station is busy with a request, that it hasn't completed and needs to make sure another node doesn't sneak another request in and cause state issues...
I'm not fully sure about the circumstances of this but perhaps due to a DCC Programming operation. The Fremo guys in Germany had made changes to the original EmbeddedLocoNet library to handle this in there LocoNet Gateway device http://dcc-mueller.de/switch/loconete.htm.

The code for this is here: https://sourceforge.net/p/embeddedloconet/svn/HEAD/tree/old-cvs/trunk/avr-base/ln_hardware_uart.c and enabled with the conditional compilation "#ifdef LOCONET_MASTER"

Anyway I'm thinking we may be able to add this needed capability by adding a new "Respond()" member for Command Station ONLY use, that handles the different CD_Backoff and OPC_BUSY semantics.

atanisoft, 2019-04-22:
> Anyway I'm thinking we may be able to add this needed capability by adding a new "Respond()" member for Command Station ONLY use, that handles the different CD_Backoff and OPC_BUSY semantics.

Perhaps have this method in a sub-class of LocoNet so that it is named as LocoNetMaster or similar? As of right now the CS code doesn't handle this behavior in an ideal way and is mostly experimental support at best (though it may work to receive throttle commands)

kiwi64ajs, 2019-04-22:
Hi Mike,

> On 23/04/2019, at 9:32 AM, Mike Dunston <notifications@github.com> wrote:
> 
> Anyway I'm thinking we may be able to add this needed capability by adding a new "Respond()" member for Command Station ONLY use, that handles the different CD_Backoff and OPC_BUSY semantics.
> 
> Perhaps have this method in a sub-class of LocoNet so that it is named as LocoNetMaster or similar? As of right now the CS code doesn't handle this behavior in an ideal way and is mostly experimental support at best (though it may work to receive throttle commands)
> 
Yeah, that’s probably a better way to handle it - just need to think through all the use-cases, but I suspect you’re right.

I need to go back and refresh myself on all that stuff, as it is all fairly stale in my mind and as I never implemented or used it myself, it was only ever second-hand. The Fremo guys did make extensive use of the LocoNet Gateway so I expect they will have nailed-it enough to work reliably in their context so with some careful diff-ing we should be able to see exactly what they did… ;)  

However, now that you’re building a command station that will also need to implement the different LocoNet behaviour, I guess its a good time to refresh myself on all this.

Are you maintaining the separate LocoNet2 library:

	https://github.com/atanisoft/LocoNet2/tree/development <https://github.com/atanisoft/LocoNet2/tree/development> 

as well as the lib/LocoNet2 code in the DCCppESP32:

	https://github.com/atanisoft/DCCppESP32/tree/master/lib/LocoNet2 <https://github.com/atanisoft/DCCppESP32/tree/master/lib/LocoNet2>

The latter looks to be ahead of the separate LocoNet2 library repo. Which should I use to experiment with?

Alex

atanisoft, 2019-04-22:
> Are you maintaining the separate LocoNet2 library: https://github.com/atanisoft/LocoNet2/tree/development <https://github.com/atanisoft/LocoNet2/tree/development> as well as the lib/LocoNet2 code in [...]

Yes, both are to be maintained actually. I need to sync them back up, a couple changes may have gone in locally in the CS tree that didn't get ported over to the standalone lib. I have a couple pending changes from another user related to the throttle code (LN throttle class) which I need to pull in and diff as well. I'll take care of syncing the two trees up over the next couple days. I'd suggest start with the CS copy and I'll sync it up over to the other.

alex-shepherd-nv, 2019-04-22:
Ok, sounds good.

You’d made some significant (but good) changes to what I had done originally getting rid of the weak referenced call-backs. They kinda seemed a "good idea” way back at that time, but are not explicit and somewhat problematic if the call-back functions are incorrectly named.

Anyway I know there were some other subtle changes in other parts of the (old/current) library code that should be put into your code.

I’ll start getting myself familiar with your codebase and see if I can get some projects working on my hardware.

I’m quite keen to produce a multi-protocol Command Station to join the LocoNet and NCE Cab-Bus worlds within Free-mo - even here in little NZ we have the two tribes contending for dominance, when I believe concurrent co-existance is quite likely possible.

Anyway time will tell…

Alex

atanisoft, 2021-12-24:
It is not feasible to support LocoNet and other protocols with ESP32 Command Station. Closing issue for now as there are alternatives available.

closed 2021-12-24 by atanisoft