atanisoft/ESP32CommandStation #8: GPIO0 does not get saved

dtworth, 2018-03-08:
It looks like the line "if(sensor->getPin() > 0) {" was added to "SensorManager::store()" .  Maybe it should be "if(sensor->getPin() >= 0) {" since GPIO0 is a valid port.

atanisoft, 2018-03-08:
While GPIO 0 is a valid pin, it has a few restrictions on its use since it is directly tied to the boot process. If you connect a sensor to this pin and it pulls the pin LOW when the esp32 is booting the boot process will be interrupted on the esp32. GPIO 0 should be avoided for inputs and should likely be rejected as a sensor pin.

atanisoft, 2018-03-08:
GPIO2 is similar in this regard.
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

dtworth, 2018-03-09:
Perhaps you should prevent that pin from being defined as an input in the first place.  The command <S 0 0 0> still works.

atanisoft, 2019-02-13:
this has been included in the development branch with plans to be included in the v1.2.0 release planned for this coming weekend.

closed 2019-02-13 by atanisoft