atanisoft/ESP32CommandStation #12: Left/Right pins instead of Enable/Direction pins

jeffeb3, 2018-05-24:
I'm using an H bridge motor controller: https://www.robotshop.com/en/l298-dual-h-bridge-dc-motor-controller.html

I am currently controlling it with an ESP32, and I'm just doing old school (DC?) PWM output on the left track or the right.

The motor controller I have doesn't have enable/direction though, it has 4 pins, two for each track, one pin for each side. It looks like in signalGeneratorPulseTimer and signalGeneratorDirectionTimer, you are digitalWrite'ing to the directionPin, and in MotorBoard, you're setting the enable pin. 

What do you think about abstracting that functionality to be compatible with other motor controllers like mine? I'm hoping to avoid #ifdefs. Maybe I should just start with a branch, finding the edits I need first, and then we can work on abstraction.

atanisoft, 2018-05-24:
The motor shield you linked to is not suitable for generating a DCC signal and does not have current sense exposed. If you are after straight DC control this code is not suitable for your needs unfortunately.

Abstraction is not appropriate here and a swap in replacement is better since it is a completely different process to control DC vs DCC. 

Feel free to fork the code and submit a pull request that provides the necessary ifdef options to switch from DCC to DC control. There would be a lot of work to make this work for standard DC.

closed 2018-05-24 by atanisoft

jeffeb3, 2018-05-25:
I don't want this to wyorl for DC control. I already have something set up for that.

I think this motor controller should work for DCC. When you set _directionPin to High, I will set pin1 high and pin2 low. When you set _directionPin low, I will do the opposite, setting pin1 low and pin2 high. How is that any different?

I've also gotten around the Rsense problem. It's not laid out, but it's a huge pin and I can easily cut it and solder in a connector instead of having it go straight to ground. I plan on adding my own Rsense pin, as well as a zener diode to protect the esp analog input. Ive testes the circuit with my DC set up.

atanisoft, 2018-05-25:
Please refer to the wiki for a solution to splitting the single pin signal to two pins. That is also how the IBT7960 motor board works.

The approach used for that board is the enable pin is used to turn on/off the board and the direction pin is passed through a transistor or not gate to split the signal into inverse values that are passed to the pins on the board. This circuit is built into the Arduino and pololu shields.