atanisoft/ESP32CommandStation #13: Pololu MC33926

crood58, 2018-09-08:
Is there any information on connecting the Pololu MC33926 to the ESP32 for DCC++? 

Thank you in advance.

atanisoft, 2018-09-10:
Unfortunately I don't have a Pololu shield to test with, I'll post details on this as soon as I have a chance to review the Pololu shield documentation and how best to connect it to the ESP32.

atanisoft, 2018-09-10:
@crood58 you can try these connections:
MOTORBOARD_ENABLE_PIN_MAIN / GPIO 25 / Pololu M1PWM & D2
MOTORBOARD_CURRENT_SENSE_MAIN / GPIO 36 (SVP) / Pololu M1FB
DCC_SIGNAL_PIN_OPERATIONS / GPIO 19 / Pololu M1DIR
MOTORBOARD_ENABLE_PIN_PROG / GPIO 23 / Pololu M2PWM
MOTORBOARD_CURRENT_SENSE_PROG / GPIO 39 (SVP) / Pololu M2FB
DCC_SIGNAL_PIN_PROGRAMMING / GPIO 18 / Pololu M2DIR

Make sure to connect GND and VDD on the Pololu to the ESP32 on GND and 3v3 pins.

atanisoft, 2019-02-13:
this has been documented and tested on the development branch with plans to be included in v1.2.0 which should be available sometime this coming weekend.

closed 2019-02-13 by atanisoft