atanisoft/ESP32CommandStation #14: Unable to connect to WiFi AP

TrainzLuvr, 2019-02-02:
I keep getting:
    `Reason: 203 - ASSOC_FAIL`
...and on and on retrying to connect with that error.

My AP is MAC locked, and I did add ESP32's MAC address into it. Does this actually pose a problem or the issue is somewhere else?

It would also be nice if the serial log displayed the MAC address for such occasions. I had to hack it in myself in the DCCppESP32.cpp :)

atanisoft, 2019-02-03:
I'm not certain what might cause this specific error. I'll dig into the esp32 code and get you some details on how to turn on more logging, but in PlatformIO.ini make sure the log level is set to verbose. That may give a bit more details but may not, I suspect a call to esp_set_log_level may be needed for the IDF code

atanisoft, 2019-02-03:
add these in WiFiInterface.cpp just above WiFi.mode(WIFI_STA):
```
  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_wifi_internal_set_log_level(WIFI_LOG_VERBOSE);
  esp_wifi_internal_set_log_mod(WIFI_LOG_MODULE_ALL, WIFI_LOG_SUBMODULE_ALL, true);
```
This will turn on a lot of extra details from the wifi module.