atanisoft/ESP32CommandStation #15: Request Integrate Withrottle (Engine Drive) backend

dlowe30971, 2019-02-12:
All, this request is to add the withrotte (Engine Driver) backend code to the ESP32 DCC++. An example is given by Vald here https://github.com/vhar/withrottle using an ESP01 and ESP8266. This would allow users to connect to the ESP32 using the Engine Driver app.

For your consideration,

atanisoft, 2019-02-12:
This has been on the to-do list for a while. With the integrated LCC support coming in v1.2.0 (release soon) this may get implemented via that instead.

dlowe30971, 2019-02-12:
Cool. Than look s Mike. Will ln look forward to testing.

One question, the on board html page, do you have a picture of that?

Dave,
dlowe30971@live.com.au
Mobile: +61 (0)432 493477

atanisoft, 2019-02-12:
Which board are you looking for?

dlowe30971, 2019-02-12:
Hey Mike, I use the ESP32 Dev boards. But I may switch to the board that fits the Arduino motor shield. I am looking to add this system to a G Scale loco.

Cheers,

atanisoft, 2019-02-12:
Ahh, yes I have one of the Arduino Uno form factor ESP32 boards. It does work decently but the pin mapping is a bit poor, the A0 and A1 pins are not usable for analog when WiFi is active so a jumper is required to A2/A3 or A4/A5.

Are you looking to have the CS setup inside the loco or powering the track/loco? If you are looking to power the track I'd suggest using a booster on the outputs from the ESP32/L298 so you are not limited to the max of ~1.75A (development branch, on master it is ~980mA).

If you are going to have the ESP32/L298 inside the loco then it should be less of a concern as the loco by itself shouldn't go beyond the 1.75A limit under normal circumstances.

dlowe30971, 2019-02-12:
Inside the loco and powered by battery either in trailing wagon or loco itself depending on space. I have already proven ESP8266 + Uno + Motor Shield + to track (HO demo). I was able to connect and number of clients and control a number of locos. But I want to save a little more space so the ESP32 is a great option.

atanisoft, 2019-02-12:
ok that makes sense, In that case the uno form factor with an L298 shield may work for your usecase, but a smaller esp32 with L298 may fit easier.