atanisoft/ESP32CommandStation #21: PCB+BOM files missing from ESP32 Command Station PCB page

TrainzLuvr, 2019-06-13:
Hi Mike,

Just noticed the updated page at https://github.com/atanisoft/ESP32CommandStation/tree/development/docs/PCB and the SVG/PDF+BOM files are missing.

I'm sure you've got them and forgot to upload, so whenever you get a chance, please and thank you.

TrainzLuvr, 2019-06-13:
I see the files are in the gh-pages branch :)

https://github.com/atanisoft/ESP32CommandStation/tree/gh-pages

atanisoft, 2019-06-13:
It will be published soon to https://atanisoft.github.io/ESP32CommandStation/ (where gh-pages is exposed). I'm still working on finalizing the design and will be ordering a test batch in the next couple days. Once I have it tested I'll be sharing on the site above as well as making available kits for purchase.

closed 2019-06-13 by atanisoft