atanisoft/ESP32CommandStation #24: Cannot build (No module named 'cStringIO')

desertspider, 2019-10-13:
Hi,

Since a autoupdate of something within PlatformIO, i can no longer build ESP32 Command Station. No matter what version.

The building stops with this error;
`ModuleNotFoundError: No module named 'cStringIO':`

Downgrading python, c++, platformio or platformio core does not solve the issue.

atanisoft, 2019-10-13:
What version are you using and which branch?

desertspider, 2019-10-13:
I tried 1.2.3, latest stable (1.3.0) and latest Dev. All same problem.

atanisoft, 2019-10-13:
Ok that would be it, python 3 removed cStringIO. I'll need to find a solution of some sort for it.

In the meantime, it may be easier to uninstall PIO, install python 2.7 (configure it as default in system path), reinstall PIO and hopefully it will use the 2.7 version. This could be a temp solution until I update the script to support both 2.x and 3.x python versions

atanisoft, 2019-10-14:
This should now be working .... The latest PIO VSCode update has moved up to Python3 by default as you noted and required a bit of rework on the resource embedding script as Python3 has removed cStringIO as a standalone module and instead moved it into the io.StringIO package but this really shouldn't be used for binary data which instead should use io.BytesIO which has also been incorporated in the above commits.

Test builds have completed from master and also on the development branch. I do not guarantee that the latest master code is 100% stable as I ran into a few issues which have been resolved more recently on the development branch only. I'm working on some remaining issues before merging development down to master for v1.5.0 release, but you should be able to use the development branch in the meantime with it being mostly stable.

closed 2019-10-14 by atanisoft

desertspider, 2019-10-14:
Thanks for the quick update. Flashing 1.5 without a problem. Only when starting the ESP32, i get this; Not sure if i should make a new issue.
Bootloop:
```
ESP32 Command Station v1.5.0 starting up (from ota_0)...
ESP32 Command Station uses the OpenMRN library
Copyright (c) 2019, OpenMRN
All rights reserved.
[NVS] Initializing NVS
[ADC] Configure 12-bit ADC resolution
E (288) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
[Config] SD Card not present or mounting failed, using SPIFFS
[Config] SPIFFS usage: 3.19/757.66 KiB
[Config] SPIFFS will be used for persistent storage.
[Config] Found existing CS config file, attempting to load...
[Config] Existing configuration successfully loaded and validated.
[LCC] Corrupt configuration file detected, /cfg/LCC/config is too small: 2147847106 bytes, expected: 482 bytes
[Turnout] Initializing DCC Turnout database
abort() was called at PC 0x400dbd90 on core 0
E (409) esp_apptrace: Application tracing via TRAX is disabled in menuconfig!

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40086d55:0x3ffc48a0 0x40086fe5:0x3ffc48c0 0x400dbd90:0x3ffc48e0 0x400e8d08:0x3ffc4920 0x400de639:0x3ffc49b0 0x400ea25b:0x3ffc4a30 0x4013ba90:0x3ffc4cf0 0x4008c2e9:0x3ffc4d10

E (433) esp_apptrace: Application tracing via TRAX is disabled in menuconfig!
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5564
load:0x40078000,len:10104
load:0x40080400,len:6504
entry 0x40080754
I (29) boot: ESP-IDF 3.30300.190916 2nd stage bootloader
I (29) boot: compile time 19:23:14
I (29) boot: Enabling RNG early entropy source...
I (31) boot: SPI Speed      : 40MHz
I (35) boot: SPI Mode       : DIO
I (38) boot: SPI Flash Size : 4MB
I (41) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (50) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (56) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (63) boot:  2 ota_0            OTA app          00 10 00010000 00190000
I (69) boot:  3 ota_1            OTA app          00 11 001a0000 00190000
I (76) boot:  4 spiffs           Unknown data     01 82 00330000 000d0000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x5fc98 (392344) map
I (226) esp_image: segment 1: paddr=0x0006fcc0 vaddr=0x3ffb0000 size=0x00350 (   848) load
I (227) esp_image: segment 2: paddr=0x00070018 vaddr=0x400d0018 size=0xd1a88 (858760) map
I (521) esp_image: segment 3: paddr=0x00141aa8 vaddr=0x3ffb0350 size=0x03100 ( 12544) load
I (526) esp_image: segment 4: paddr=0x00144bb0 vaddr=0x40080000 size=0x00400 (  1024) load
I (527) esp_image: segment 5: paddr=0x00144fb8 vaddr=0x40080400 size=0x10974 ( 67956) load
I (570) boot: Loaded app from partition at offset 0x10000
I (570) boot: Disabling RNG early entropy source...

```

atanisoft, 2019-10-14:
@desertspider Start with erase flash from PIO and reupload the code, it looks like there is a corruption in the SPIFFS partition that is not being handled correctly. If that fails please decode the backtrace line using [this](https://github.com/me21/EspArduinoExceptionDecoder) decoder with a commandline similar to:
```
decoder.py -p ESP32 -e .pio\build\esp32\firmware.elf stack.txt -t $HOMEDRIVE$HOMEPATH\.platformio\packages\toolchain-xtensa32
```
with the backtrace in ```stack.txt```

desertspider, 2019-10-14:
Erase flash did fix it. 

On an side note. The reason i re-flashed the latest version was because i didn't get 1.3 to work. I did manage to get the same shield to work with dcc++ on a mega. 
Problem is, ESP32CommandStation turns the track off when i send a command to the train. Tried with 2 trains. Both has a different decoder brand. How can i debug this?

atanisoft, 2019-10-14:
> Erase flash did fix it.

Great, I'll setup a v1.2.3 build and "upgrade" to the v1.5 version and see what is causing it to go awry.

> Problem is, ESP32CommandStation turns the track off when i send a command to the train. Tried with 2 trains. Both has a different decoder brand. How can i debug this?

Sounds like the CS is crashing and restarting. Check the console output for the cause, it should have a backtrace like you posted above and you should be able to decode it using the py file above. Open a new issue to track that failure and include the config modules that you have enabled etc. I'll take a look at it more closely and see what might be causing it.