atanisoft/ESP32CommandStation #27: Building and installing with Visual Studio

buelowp, 2019-11-28:
Hi Mike, I'm getting back into this after a long break. I think this is the right repository now, I see that others have been obsoleted. However, I see the documentation, and a how to build, but it doesn't actually tell how to actually do this with with VSCode. I use VSCode for Particle, but for the ESP32, I seem to be striking out. I was wondering if you might update the git pages sections to indicate how to build and how to flash using VSCode. I'll keep working on it myself in the meantime, but it would be nice to round out that portion of the documentation.

atanisoft, 2019-11-28:
Sorry about that, the docs are a bit sparse in a few areas. In VSCode, install the PlatformIO IDE extension as first step. Add the esp32 CS directory to the workspace. Edit the include/Config_xxx.h files, when done editing these keep one of them open. Click the PIO icon (alien on left sidebar). Plugin your esp32 module via usb. Click on upload and monitor. PlatformIO should take care of the rest for you.

atanisoft, 2019-11-28:
I'll add more extensive documentation soon.

buelowp, 2019-11-28:
Thanks, I assumed it was simple, but I haven't gotten it going yet. I appreciate it.

buelowp, 2019-11-28:
Thanks, that got me over the hump. I wasn't familiar with the PIO plugin and missed the icon and what it could do.

atanisoft, 2019-11-28:
Keeping this open to track the doc updates... More on that after this coming weekend.

Glad that you were able to get it up and going. There are a few known issues with the current master code and possibly some with the web interface due to the GitHub rawgit CDN being shutdown. Almost all current work is done on the development branch and is mostly stable, there are a few issues which I'm still investigating though

buelowp, 2019-11-28:
Perfect, thanks. No rush. I'm using our existing solution until after the holiday season is over when I will have time to do prototyping with the ESP32 setup before we put everything away for the season. I was hoping to get a head start and play with some pieces on the side during work breaks. We'll see how well MTH does DCC for O gauge.

atanisoft, 2020-10-16:
Closing this as the build process has been shifted to using idf.py (or VSCode IDF extension) and is now documented on the web page.

closed 2020-10-16 by atanisoft