atanisoft/ESP32CommandStation #28: DCC MP1 turnouts not working

zubozrout, 2020-01-11:
Hi,
sorry but I am wondering what's needed in order to make turnouts work?

I am using these devices: http://www.fucik.name/masinky/MP1/
but it seems no matter what I do the device doesn't try to switch the motor in the MTB MP1 machine.

I have been successfully using these devices with https://github.com/DccPlusPlus/BaseStation previously but now that I've switched over to ESP32CommandStation this doesn't work.

And it seems to make no difference if I try to switch the switches with the web based version of ESP32CommandStation or through JMRI.

I would appreciate any thought on this, thank you.

zubozrout, 2020-01-11:
I've also just tested it with https://www.espb.cz/DCC-master-modul-pro-unikatni-prestavniky-H0-TT-N-espb-cz-d239.htm (am not normally using this one, but have one with me) and that doesn't work either :(.

atanisoft, 2020-01-11:
@zubozrout Some question...
1. How are you creating the Turnout in ESP32 CS, are you creating it via DCC address or by board/index (DCC++ style)?
2. Which version of ESP32 CS?
3. Can you capture the serial console from the ESP32 to ensure it is receiving/processing the commands?

zubozrout, 2020-01-11:
Hi, thank you for the answer.
Well, I've been simply trying to switch the turnout using:
Command Station → Turnouts → + → Address Mode: DCC address only → Address = 1
(similarly in JMRI using a Turnout control https://www.jmri.org/manual/JMRI_2-12_Manual/Main_TurnoutControl.shtml)

I think the version I am currently having is `iDCC++ ESP32 Command Station: V-1.3.0 / Aug 20 2019 19:32:04` - which matches the repo version I have on my local computer, that's master `v1.3.0-rc2-6-gac3af3f`.

I've got the following capture from browser's web console when trying to switch the turnout:
```
WS: <H 1 1>
WS: <H 1 0>
WS: <H 1 1>
WS: <H 1 0>
WS: <H 1 1>
WS: <H 1 0>
```

But that's probably not that much useful, and here is a serial output from the device itself:
[ESP32CommandStation-turnout.log](https://github.com/atanisoft/ESP32CommandStation/files/4049536/ESP32CommandStation-turnout.log)

If I try to access the turnout on address 1:
`[Turnout 1] Created using DCC address 1 as type LEFT and initial state of Closed
`
, I get this:
`[Turnout 1] Unable to set state, turnout not found`
while in other cases it just seem not do to anything at all.

And when I connect two pins on the decoder I am using http://www.fucik.name/masinky/MP1/ with a jumper it should change its address respectively and listen to anything. I was trying to do that but also have tried calling the decoders with their current addresses with no luck either.

atanisoft, 2020-01-11:
```
[OPS] RMT feeder task starting up
[Turnout 1] Created using DCC address 1 as type LEFT and initial state of Closed
Incoming connection from 192.168.1.29, fd 55.
[W][AsyncTCP.cpp:737] _poll(): rx timeout 4
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
....
```
It appears the CS crashed soon after the turnout was created. This is why you were seeing an issue with turnout address 1 not responding. However, there are a few issues on the v1.3.0 code which prevented the release from going out. I would suggest downgrade to https://github.com/atanisoft/ESP32CommandStation/releases/tag/v1.2.3 as that is the last stable release. If you prefer to try the bleeding edge you can try the development branch (v1.5.0). In either case I would highly suggest erasing flash to start fresh.

zubozrout, 2020-01-11:
Thank you very much @atanisoft. I'll try the bleeding edge then or downgrading in case that's not going to work and then try again :). Will report back with the results once I try.

zubozrout, 2020-01-11:
Ok, so it seems v1.2.3 was the way to go as the switches seem to work with it just fine - thank you :).
Unfortunately the bleeding edge devel release wasn't able to power up the track despite the motor shield lighting up on power turned on in the UI, but it seems no power was actually going on to the tracks as it didn't even lighten up any LEDs connected to it (and my multi-meter didn't measure anything there either) - so I wasn't able to test anything with it.

Anyway, since v1.2.3 is the latest stable I would either suggest to close this issue as not relevant anymore or maybe keep this open until devel is more ready to be tested and then I can retry. Not sure.

atanisoft, 2020-01-11:
Ok, glad that it works with v1.2.3. I'll do some further testing with the development branch. Were you using the L298 motor shield?

zubozrout, 2020-01-11:
Thank you and yes, one of the cheap copies, but yes, it is the L298 based one.

atanisoft, 2020-01-12:
Ok, I have a Deek-Robot one that I'll do some further testing with. From my scope testing the signal is being generated correctly so it is going to be something simple likely.

atanisoft, 2020-09-21:
@zubozrout Can you test the v1.5.0 alpha3 release? It should work with your decoders now.

closed 2020-09-21 by atanisoft

zubozrout, 2020-09-21:
Ah, sorry @atanisoft, I completely forgot I also opened up this issue as I've then moved to the TrainBoard forum where I felt it was easier. Anyway, it all works great for me since at least since [July 21](https://www.trainboard.com/highball/index.php?threads/esp32-command-station.112634/page-52#post-1144386), so thank you very much again, really appreciate it :). Also thank you answering and closing this ticket ... all resolved now.