atanisoft/ESP32CommandStation #29: index_html.h

arnold-b, 2020-02-19:
Hello, 
WOuld be more easy to publish index_html.h not everyone is using platform.io 
or at leased should be published the version of python. 

Import("env") dont exist

atanisoft, 2020-02-19:
The last release v1.2.3 will not work in PIO due to Python version updates by PIO. However, the usage of PIO is not going to be required starting with v1.5.0. v1.5.0 will be using ESP-IDF v4 directly for the build system which is currently not supported by PIO and is build via the standard ESP-IDF approach instead. There will also be pre-built binaries for the PCB kit which will soon be released as well.

atanisoft, 2020-02-20:
@arnold-b For the current PIO release you can replace the build_index_header.py file with the content below and it will work as expected:
```
#######################################################################
# ESP32 COMMAND STATION
#
# COPYRIGHT (c) 2017-2019 Mike Dunston
#
#  This program is free software: you can redistribute it and/or modify
#  it under the terms of the GNU General Public License as published by
#  the Free Software Foundation, either version 3 of the License, or
#  (at your option) any later version.
#  This program is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#  GNU General Public License for more details.
#  You should have received a copy of the GNU General Public License
#  along with this program.  If not, see http://www.gnu.org/licenses
#######################################################################

Import("env")
import gzip
import os
import struct
from io import BytesIO
import shutil

def build_index_html_h(source, target, env):
    if os.path.exists('%s/include/index_html.h' % env.subst('$PROJECT_DIR')):
        if os.path.getmtime('%s/data/index.html' % env.subst('$PROJECT_DIR')) < os.path.getmtime('%s/include/index_html.h' % env.subst('$PROJECT_DIR')):
            return
    print("Attempting to compress %s/data/index.html" % env.subst('$PROJECT_DIR'))
    gzFile = BytesIO()
    with open('%s/data/index.html' % env.subst('$PROJECT_DIR'), 'rb') as f, gzip.GzipFile(mode='wb', fileobj=gzFile) as gz:
        shutil.copyfileobj(f, gz)
    gzFile.seek(0, os.SEEK_END)
    gzLen = gzFile.tell()
    gzFile.seek(0, os.SEEK_SET)
    print('Compressed index.html.gz file is %d bytes' % gzLen)
    with open('%s/include/index_html.h' % env.subst('$PROJECT_DIR'), 'w') as f:
        f.write("#pragma once\n")
        f.write("const size_t indexHtmlGz_size = {};\n".format(gzLen))
        f.write("const uint8_t indexHtmlGz[] PROGMEM = {\n");
        while True:
            block = gzFile.read(16)
            if len(block) < 16:
                if len(block):
                    f.write("\t")
                    for b in block:
                        # Python 2/3 compat
                        if type(b) is str:
                            b = ord(b)
                        f.write("0x{:02X}, ".format(b))
                    f.write("\n")
                break
            f.write("\t0x{:02X}, 0x{:02X}, 0x{:02X}, 0x{:02X}, "
                    "0x{:02X}, 0x{:02X}, 0x{:02X}, 0x{:02X}, "
                    "0x{:02X}, 0x{:02X}, 0x{:02X}, 0x{:02X}, "
                    "0x{:02X}, 0x{:02X}, 0x{:02X}, 0x{:02X},\n"
                    .format(*struct.unpack("BBBBBBBBBBBBBBBB", block)))
        f.write("};\n")

env.AddPreAction('$BUILD_DIR/src/WebServer.cpp.o', build_index_html_h)
```

Also note that this will only run within the context of PIO and no other build environments are supported. As mentioned above v1.5.0 will not support PIO for building unless PIO moves to supporting CMake based build systems.

atanisoft, 2020-05-30:
The py file that was causing problems with PlatformIO has been removed as of the v1.5.0-alpha1 release.

closed 2020-05-30 by atanisoft