atanisoft/ESP32CommandStation #30: Problem to compile.

arnold-b, 2020-02-19:
Hello 
What is the used Version of ESPAsyncWebServer ?

Thanks

arnold-b, 2020-02-19:
Found Result: 

ESPAsync is also using the file webserver.x 

Compiler is only using the project one and not this one. 
Solution. rename project webserver.x to something else

closed 2020-02-19 by arnold-b

atanisoft, 2020-02-19:
The issue with ESPAsyncWebServer is due to a recent PIO update which broke LDF. Edit the platformio.ini file to ensure [these two lines](https://github.com/atanisoft/ESP32CommandStation/blob/master/platformio.ini#L35-L36) are present and not the original [two](https://github.com/atanisoft/ESP32CommandStation/blob/9c95a8d4f480d6a42c6112187ef9edce822c1a19/platformio.ini#L33-L34).