atanisoft/ESP32CommandStation #40: Reconfigure wifi

jpnarkinsky, 2020-07-21:
Hi there, I'm trying to get this built and running for the first time.  I configured the wrong wifi SSN (the 5.8 ghz one instead of the 2.4 ghz one) and flashed it to my device.  I figure this out quickly, and I've reconfigured it, but it continues to try to connect the 5.8 ghz SSN.  That is, the changes to the wifi configuration don't seem to be taking effect.  I've done a `idf.py fullclean`, etc., but nothing seems to be taking effect.

Any suggestions?  I'm brand new to ESP32, so it's certainly possible I'm doing it wrong.  Thanks for any suggestions!

atanisoft, 2020-07-21:
Run: `idf.py erase_flash` and then `idf.py flash monitor`.

atanisoft, 2020-07-21:
Another option is enable the factory reset pin via config and flash that with the pin held low. It should force clear settings and reconfigure based on the values you enter in the config.

jpnarkinsky, 2020-07-21:
Thanks!  That did it.

closed 2020-07-21 by jpnarkinsky