atanisoft/ESP32CommandStation #42: Add pictures of supported motor shields to webpage

atanisoft, 2020-08-18:
Currently the webpage only lists supported motor shields but without pictures, in the case of the L298 boards there are multiple variants available and not usable without modifications (red PCB with vertical L298 lacks current sense)

TrainzLuvr, 2020-08-18:
This one: https://i.ibb.co/80GGSvT/L298-N-board.jpg ?

Sense A/B on it right now connects to the ground, but it could be modified by cutting the traces and inserting resistors in appropriate places.

atanisoft, 2020-08-18:
Yup, that would be one of the L298 boards that requires modification for the current sense lines. I have a couple of these (or a close variant) and need to document the modifications.

> it could be modified by cutting the traces and inserting resistors in appropriate places.

Yes, those are the modifications that are required. The two current sense pins on the L298 need to be cut to create a voltage divider circuit with the output going to the analog input of the esp32.

TrainzLuvr, 2020-08-18:
Ok, so we were thinking the same thing.

I have not done it yet myself, because in all honesty, why even use L298 anymore when there are so many better alternatives. :)

atanisoft, 2020-08-18:
> why even use L298 anymore when there are so many better alternatives.

Very true but there are a lot of people purchasing these sort of boards for cost reasons without knowing they do not work out-of-the-box.

michelcve, 2021-02-24:
@atanisoft Perhaps having a look at what DCC++ EX did, might help setting up such a page?

https://dcc-ex.com/reference/hardware/motor-boards.html

atanisoft, 2021-02-24:
@michelcve That is very similar to what I was planning but have not done yet. Almost any of the motor shields that DCC++EX supports should work on ESP32 CS as well, but they may require customization of the config etc.

closed 2022-02-08 by atanisoft