atanisoft/ESP32CommandStation #49: Latest master with ESP-IDF v4.1 - ldgen fails to create esp-idf/esp32/ld/esp32.project.ld

TrainzLuvr, 2020-09-03:
Got another error:

`[7/32] Generating ld/esp32.project.ld
FAILED: esp-idf/esp32/ld/esp32.project.ld
cd /mnt/r/CODE/ESP32CommandStation/build/esp-idf/esp32 && /home/user/esp/espressif/python_env/idf4.1_py3.7_env/bin/python /home/user/esp/esp-idf/tools/ldgen/ldgen.py --config /mnt/r/CODE/ESP32CommandStation/sdkconfig --fragments /home/user/esp/esp-idf/components/xtensa/linker.lf   /home/user/esp/esp-idf/components/esp_ringbuf/linker.lf /home/user/esp/esp-idf/components/spi_flash/linker.lf     /home/user/esp/esp-idf/components/espcoredump/linker.lf /home/user/esp/esp-idf/components/esp32/linker.lf       /home/user/esp/esp-idf/components/esp32/ld/esp32_fragments.lf     /home/user/esp/esp-idf/components/soc/linker.lf /home/user/esp/esp-idf/components/esp_event/linker.lf   /home/user/esp/esp-idf/components/esp_wifi/linker.lf      /home/user/esp/esp-idf/components/lwip/linker.lf        /home/user/esp/esp-idf/components/log/linker.lf /home/user/esp/esp-idf/components/heap/linker.lf  /home/user/esp/esp-idf/components/freertos/linker.lf    /home/user/esp/esp-idf/components/newlib/newlib.lf      /home/user/esp/esp-idf/components/app_trace/linker.lf     /home/user/esp/esp-idf/components/esp_gdbstub/linker.lf --input /home/user/esp/esp-idf/components/esp32/ld/esp32.project.ld.in --output /mnt/r/CODE/ESP32CommandStation/build/esp-idf/esp32/ld/esp32.project.ld --kconfig /home/user/esp/esp-idf/Kconfig --env-file /mnt/r/CODE/ESP32CommandStation/build/config.env --libraries-file /mnt/r/CODE/ESP32CommandStation/build/ldgen_libraries --objdump /home/user/esp/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/home/user/esp/esp-idf/tools/ldgen/ldgen.py", line 158, in <module>
    main()
  File "/home/user/esp/esp-idf/tools/ldgen/ldgen.py", line 134, in main
    mapping_rules = generation_model.generate_rules(sections_infos)
  File "/home/user/esp/esp-idf/tools/ldgen/generation.py", line 352, in generate_rules
    self._create_exclusions(mapping_rules, default_rules, sections_infos)
  File "/home/user/esp/esp-idf/tools/ldgen/generation.py", line 451, in _create_exclusions
    general_rule.add_exclusion(specific_rule, sections_info)
  File "/home/user/esp/esp-idf/tools/ldgen/generation.py", line 100, in add_exclusion
    expansions = do_section_expansion(self, section)
  File "/home/user/esp/esp-idf/tools/ldgen/generation.py", line 84, in do_section_expansion
    expansions = fnmatch.filter(sections_in_obj, section)
  File "/usr/lib/python3.7/fnmatch.py", line 55, in filter
    for name in names:
TypeError: 'NoneType' object is not iterable
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

This seems to relate to https://github.com/espressif/esp-idf/issues/5734 but I kinda agree with m-bruck - why compile a driver in, for something that's not being used. ¯\_(ツ)_/¯

CONFIG_LWIP_IRAM_OPTIMIZATION is enabled for ESP32CS build.

atanisoft, 2020-09-03:
Yeah, this one is a nasty one, I worked around it locally in generation.py to avoid the None case. I'll submit a PR to upstream to fix it.

closed 2020-09-03 by atanisoft

TrainzLuvr, 2020-09-03:
Sorry to re-open this but the error still persist. I pulled the latest master few minutes ago.

Running ESP-IDF v4.1.

atanisoft, 2020-09-03:
Did you remove the sdkconfig file? If not you will need to comment out `CONFIG_LWIP_IRAM_OPTIMIZATION` which is disabled [here](https://github.com/atanisoft/ESP32CommandStation/blob/master/sdkconfig.defaults#L36-L37).

TrainzLuvr, 2020-09-03:
That did it, my sdkconfig still had ...=y

Compiled to completion now. Thanks!

karlp, 2021-03-09:
I'm still getting this on 4.2 release, with LWIP_IRAM_OPTIMIZATION unchecked in menuconfig?

atanisoft, 2021-03-09:
> I'm still getting this on 4.2 release, with LWIP_IRAM_OPTIMIZATION unchecked in menuconfig?

Can you send me your sdkconfig and I'll compare it to the one I'm using and see if I can find which option is causing the issue.

karlp, 2021-03-09:
Sure, I've attached it here.  I suspect it might be wonky toolchains though.  I get this if I run cmake directly inside netbeans, from a plain project (no prior build, ie akin to a clean checkout of the project from git)

If I run "idf.py build" once, it works, so... "unsupported tooling" I guess? 
[sdkconfig.txt](https://github.com/atanisoft/ESP32CommandStation/files/6109153/sdkconfig.txt)

also.... I'm sorry, I missed that I wason't on the repo I thought I was on, so this isn't really your issue at all, I just got here from a bug report on esp-idf, sorry!

atanisoft, 2021-03-09:
> also.... I'm sorry, I missed that I wason't on the repo I thought I was on, so this isn't really your issue at all, I just got here from a bug report on esp-idf, sorry!

No worries. Try disabling the following options:
> CONFIG_ESP32_WIFI_IRAM_OPT=y
> CONFIG_ESP32_WIFI_RX_IRAM_OPT=y

karlp, 2021-03-11:
hrm, that didn't help, but it still works with "idf.py build" so something's going odd with how cmake runs from inside netbeans.  Once it's built once, even a clean/build with netbeans works.  

thanks for looking at this, but i'm not going to follow it up further at this point, building once from the commandline is ok for me rightnow.

atanisoft, 2021-03-11:
> Once it's built once, even a clean/build with netbeans works.

The problem is very likely how netbeans is executing the ld generation then. There was a valid bug in ldgen.py but only if it encountered a file that did not have the type in it that was expected. It was a very specific pattern and it is entirely possible that netbeans is using an older copy of the files that did not have the fix included.