atanisoft/ESP32CommandStation #53: Bogus currents values reported (dived by 1000)

daniviga, 2020-09-13:
First of all I would thank you for this great project, it's really a nice way to learn an play with DCC and the ESP32.

I'm however facing an issue with current readings/reportings.

### Setup
- WeMos D1 R32
- Deek-Robot Motorshield (L298P, Arduino clone), A0 to A4 and A1 to A5, brake and VIN pins cut
- IDF 4.1
- Commit c52c483c61801dd48c22efe2ac033279c716be46
- Pretty standard setup:
  - GPIO input and sensors disabled (`CONFIG_GPIO_OUTPUTS` and `CONFIG_GPIO_SENSORS`), I'm not using them
  - SSID mode
  - SNTP enabled and TZ to Europe/Rome

OPS is working fine, I can command my DCC engines (H0n3, H0n30 and 0n30), however I always get `[OPS]      0 mA / 2000 mA` even when there's a load (engine on track running); shortages are properly spotted: `[OPS] Shutdown threshold breached   2.00 mA (raw: 4095 / 4080)`, but only 2.00 mA are reported instead of 2000 mA.

Digging into the code I've found the following:

https://github.com/atanisoft/ESP32CommandStation/blob/c52c483c61801dd48c22efe2ac033279c716be46/components/DCCSignalGenerator/private_include/MonitoredHBridge.h#L95-L102

That formula actually returns a value that looks reasonable for mA, i.e. `((50 * 2000) / 4096 =      24)`

However, when `getUsage()` is called, the value returned is dived by 1000, like if it was returning Amps and not mA:
https://github.com/atanisoft/ESP32CommandStation/blob/c52c483c61801dd48c22efe2ac033279c716be46/components/DCCSignalGenerator/MonitoredHBridge.cpp#L263-L265

As a quick and dirty test I've changed `getUsage()` to return A and not mA (multiplying by 1000, so I can change only one point in the code) and now returned values make sense:

```c
return ((lastReading_ * maxMilliAmps_) / 4096.0f * 1000.0f);
```

I get back

```
[OPS]      7 mA / 2000 mA     # (15 * 2000 / 4096=      7) 
[OPS] Shutdown threshold breached 1999.51 mA (raw: 4095 / 4080)
```

Is this a bug or it's a peculiarity of my setup (maybe the L298)? I'm also unable to read CV on PROG, not sure if this could be related or it's a different issue.

As a reference the same Shield works fine (readings too) with an Arduino Uno R3 and the BaseStation-EX (via JMRI).

atanisoft, 2020-09-13:
Thanks for the detailed report and after a quick review I think your analysis is correct. It is incorrectly scaling to Amps rather than tracking fully in mA.

For the PROG track, can you check the serial output and ensure it is not timing out or throwing another type of error? I'll do some digging on this and see if I can reproduce it as well.

atanisoft, 2020-09-14:
The commit should fix the readings but keeping this open until we narrow down the issues with the PROG track.

atanisoft, 2021-12-24:
I can no longer reproduce this issue with v2.0 code.

closed 2021-12-24 by atanisoft