atanisoft/ESP32CommandStation #60: OPS on/off button missing

TrainzLuvr, 2020-09-18:
[https://i.ibb.co/q0yCCFc/Capture.jpg](https://i.ibb.co/q0yCCFc/Capture.jpg)

atanisoft, 2020-09-19:
@TrainzLuvr This appears to be due to a cached copy of the index.html file, either in the build process or in the browser. The json payload was corrected in https://github.com/atanisoft/ESP32CommandStation/commit/6e54a6b8ebca5a3aaac3cb7b7f55973d2da115c8#diff-b7b46c8bc3f1e795a74319b4a3a9e9dd and the index.html fixed in https://github.com/atanisoft/ESP32CommandStation/commit/2e4a5eac74d5b5ecc84509ae14cbb8523bf281b9#diff-70b6324729a9676426e116cb174c0638.

I'd suggest delete build/index.html.gz.S and data/index.html.gz and rebuild/flash. That will leave just the browser which has an easy workaround, hold control key and click refresh and it should ignore the local cache and request the files again from the CS.

I'll keep this open and do some further testing on the caching as it may not be working correctly.

atanisoft, 2020-09-19:
There is a bug in the `Last-Modified` field which is causing part of this problem, that issue is tracked via https://github.com/atanisoft/HttpServer/issues/3

closed 2020-09-19 by atanisoft