atanisoft/ESP32CommandStation #69: Unable to turn off Uplink reconnect and uplink IP

ggee, 2020-09-23:
I am unable to turn off the LCC WiFi Uplink 'Uplink reconnect'.  When I turn it off and click save, the slider turns back on and strangely the 'Connect to Network' button appears.  I also constantly see the following message in the serial monitor.

[Uplink] Reconnecting to 192.168.10.170:12021.
connect: Software caused connection abort

There seems to be no way to forget the uplink settings.

TrainzLuvr, 2020-09-23:
I can confirm that's happening here as well.

Uplink reconnect goes back to On position

ggee, 2020-09-23:
I had to do a Factory Reset to clear the uplink IP.  Not sure if it was related, but when the problem was occurring, it was difficult to keep a stable DCC++ Ethernet connection from JMRI.  The response was very slow and sometimes seem to take 30 seconds for the command to be executed to just stop.  The power status would occasionally report as Unknown.

TrainzLuvr, 2020-09-23:
Do you need to add the uplink manual host value though?

For me that field is empty, as I have the mode setup to Auto, Manual, and mDNS finds JMRI no problem via the LCC network.

ggee, 2020-09-23:
Yes, the field is empty for me as well. I'm not sure what I clicked to get the CS to try and connect to JRMI constantly.

atanisoft, 2020-09-23:
@ggee the CS tracks the last known address that it successfully connected to and will by default try and reconnect to it. If that connection fails though it doesn't fallback to trying to find a new address. I'll add an option to clear the last address.

atanisoft, 2020-09-23:
@ggee can you test the update in https://github.com/atanisoft/ESP32CommandStation/pull/71? If you used `git clone` to setup your local copy of the CS code you can run `git checkout uplink` and it should pull in the changes from that PR and you can then run `idf.py build flash monitor`

ggee, 2020-10-21:
I'm running into a problem.  I entered an IP and port in the LCC Uplink manual host/port, then I saved.  Then I deleted the two values and pressed save.  This caused the node to crash.

mDNS Service Name: blank
LCC Uplink manual host: blank
LCC Uplink manual port: blank
Uplink reconnect: On

Click Save

```
Incoming connection from 192.168.10.170, fd 53.
[Uplink] Disconnecting from uplink.
[WiFi] Setting TX power to: 78
[Uplink] Reconnecting to 192.168.10.170:12021.
[Uplink] Connected to hub, configuring GridConnect port.
Allocating new alias B89 for node 05020103f000
Incoming connection from 192.168.10.170, fd 53.
[TaskMon] uptime: 00:05:56 freeHeap: 140716, largest free block: 70820, tasks: 21, mainBufferPool: 0.95kB
Incoming connection from 192.168.10.170, fd 53.
abort() was called at PC 0x400d66e3 on core 1
0x400d66e3: __cxa_end_catch at C:/Users/ggee/Desktop/esp-idf/components/cxx/cxx_exception_stubs.cpp:13

ELF file SHA256: 2cde1b2b81f68a5e

Backtrace: 0x400873b0:0x3ffc74e0 0x400877a5:0x3ffc7500 0x400d66e3:0x3ffc7520 0x40196e39:0x3ffc7540 0x4010f005:0x3ffc7560 0x400edd4f:0x3ffc7590 0x401a8b27:0x3ffc76a0 0x40117157:0x3ffc76c0 0x4011b7fd:0x3ffc7710 0x4011b609:0x3ffc7740 0x401a9679:0x3ffc7770 0x40126f6f:0x3ffc7790
0x400873b0: invoke_abort at C:/Users/ggee/Desktop/esp-idf/components/esp32/panic.c:157

0x400877a5: abort at C:/Users/ggee/Desktop/esp-idf/components/esp32/panic.c:172

0x400d66e3: __cxa_end_catch at C:/Users/ggee/Desktop/esp-idf/components/cxx/cxx_exception_stubs.cpp:13

0x40196e39: std::__throw_invalid_argument(char const*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:74

0x4010f005: http::HttpRequest::param(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int) at c:\users\ggee\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\ext/string_conversions.h:83
 (inlined by) ?? at c:\users\ggee\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/basic_string.h:6411
 (inlined by) http::HttpRequest::param(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int) at c:\users\ggee\git\esp32commandstation\build/../components/HttpServer/src/HttpRequest.cpp:273

0x400edd4f: process_config(http::HttpRequest*) at c:\users\ggee\git\esp32commandstation\build/../main/WebServer.cpp:624 (discriminator 2)

0x401a8b27: std::_Function_handler<http::AbstractHttpResponse* (http::HttpRequest*), http::AbstractHttpResponse* (*)(http::HttpRequest*)>::_M_invoke(std::_Any_data const&, http::HttpRequest*&&) at c:\users\ggee\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:282

0x40117157: http::HttpRequestFlow::process_request_handler() at c:\users\ggee\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
 (inlined by) http::HttpRequestFlow::process_request_handler() at c:\users\ggee\git\esp32commandstation\build/../components/HttpServer/src/HttpRequestFlow.cpp:395

0x4011b7fd: StateFlowBase::run() at c:\users\ggee\git\esp32commandstation\build/../components/OpenMRNLite/src/executor/StateFlow.cpp:63 (discriminator 4)

0x4011b609: ExecutorBase::entry() at c:\users\ggee\git\esp32commandstation\build/../components/OpenMRNLite/src/executor/Executor.cpp:324

0x401a9679: OSThread::start(void*) at c:\users\ggee\git\esp32commandstation\build/../components/OpenMRNLite/src/os/OS.hxx:193

0x40126f6f: os_thread_start at c:\users\ggee\git\esp32commandstation\build/../components/OpenMRNLite/src/os/os.c:391

Rebooting...
```

atanisoft, 2020-10-21:
I'll take a look into it. It should have persisted the config but may not have flushed to disk when it crashed.

atanisoft, 2020-10-21:
I've merged PR #71 to master but not closed this as I'm going to dig into what went wrong with the above exception on my desk before closing this. I'd also like to offload the uplink creation if I can.

atanisoft, 2020-10-24:
@ggee I've put your crash dump in https://github.com/atanisoft/HttpServer/issues/7 for tracking and applied a workaround for blank parameters. This fix will come in via https://github.com/atanisoft/ESP32CommandStation/pull/90

closed 2021-12-24 by atanisoft