atanisoft/ESP32CommandStation #70: Update the README.md for better grasp

keivalya, 2020-09-23:
I notice the README.md is not the best that it could potentially be. I can be more elaborative and easy to read for even beginners.
It can also be made more visually attractive.

keivalya, 2020-09-23:
I'g like to help in doing so.

atanisoft, 2020-09-23:
@keivalya I'm not sure I understand what you are intending. The README.md file is intentionally brief as the documentation is on the website instead.  But feel free to submit a PR with your updates and I'll take a look

keivalya, 2020-09-23:
Definitely!

atanisoft, 2020-10-16:
@keivalya Any progress on a PR?

keivalya, 2020-10-16:
No @atanisoft 
I think you were right. README is already enough elaborative for a beginner.

You may close this issue. I won't be able to work on it even if I could, due to my exams.
Sorry for the trouble. I should have followed up.

atanisoft, 2020-10-16:
@keivalya No worries, life can be very busy at times. Best of luck in your exams!

closed 2020-10-16 by atanisoft

keivalya, 2020-10-16:
Thank you so much for understanding @atanisoft 😄