atanisoft/ESP32CommandStation #88: CDI: Missing ability to control OPS track power. All other options should be added as well...

TrainzLuvr, 2020-10-20:
There's no button in the JMRI/CDI to create an event to turn the OPS track power on/off.

Kinda bummer if you want to control something like that from a panel and not a WebUI.

For that matter, all of the rest of the options should be added (Turnouts, Sensors...)

atanisoft, 2020-10-20:
Jmri doesn't expose the option today but you can control on/off state of OPS track by send events to the bus. Similarly estop can be sent.

This will need to be enhanced on the jmri side. For panels, you could have them fire the events.

atanisoft, 2020-10-20:
well-known events that control the areas you are after:
- turn OPS track off: [01.00.00.00.00.00.FF.FF](https://github.com/openlcb/documents/blob/master/standards/generated/EventIdentifiersS.txt#L61)
- turn OPS track on: [01.00.00.00.00.00.FF.FE](https://github.com/openlcb/documents/blob/master/standards/generated/EventIdentifiersS.txt#L63)
- e-stop (all): [01.00.00.00.00.00.FF.FD](https://github.com/openlcb/documents/blob/master/standards/generated/EventIdentifiersS.txt#L65)
- e-stop (all) clear: [01.00.00.00.00.00.FF.FC](https://github.com/openlcb/documents/blob/master/standards/generated/EventIdentifiersS.txt#L67)

The current iteration of sensors will not be added to the CDI as it doesn't make sense due to limited number of IO pins available. S88 Sensors may be added to the CDI at some point in the future but that can be tracked via it's own issue.

Turnouts can be controlled via DCC address or via LCC events:
- [01.01.02.00.00.FF.<12-bit DCC address>](https://github.com/openlcb/documents/blob/master/drafts/generated/EventIdentifiersS.txt#L102-L107) -- activate
- [01.01.02.00.00.FE.<12-bit DCC address>](https://github.com/openlcb/documents/blob/master/drafts/generated/EventIdentifiersS.txt#L108-L113)  -- deactivate
Note: turnout control via events is not currently part of the adopted standard but is supported/implemented in various libraries (including OpenMRN) and has been for quite some time.

Other well-known events that may be generated by the CS:
- [01.00.00.00.00.00.FF.F1](https://github.com/openlcb/documents/blob/master/drafts/generated/EventIdentifiersS.txt#L64-L65) -- brownout detected (not yet implemented)