atanisoft/ESP32CommandStation #89: ESP32 'Espduino' modules - very low TX WiFi power issues

roadsnail, 2020-10-22:
While carrying out some testing, I noticed that on occasions my ESP32 CS took a while to connect to my test WiFi with retries observed on the serial output.

Checking the status of my CS (as seen by the AP), I noticed the WiFi signal received by the AP was very low. Approx -85dBm, sometimes worse! Putting the CS into STA and AP mode made no difference, with the TX still low for AP and receive signal at my test AP. I independently checked the signal level with WiFi Analyser on my phone.

Thinking I may have a faulty module, I ordered up a new ESPduino (ESP32 on arduino form factor module just to be clear) and the results are exactly the same having flashed yesterday's master branch of CS.

Interestingly, but confusingly, a different module (NodeMCU ESP-32) does not seem to have an issue with TX WiFi power! The signal received from this module at my test AP is in the right ballpark given distance (around -50dBm):  a huge difference of around 35dBm when compared to the ESPDuino modules (both of them).

In order to make sure the suspect modules are actually capable of higher transmit power and I haven't somehow broken them, I decided to do a full erase in Espressif's download tool and then flash micropython for ESP32 with the same tool. After flashing, micropython connects to my test AP without issue and at a sensible TX power level. (Around -50dBm) for both 'suspect' modules, thus proving the modules are indeed capable of operating at a higher TX WiFi power level and are not broken.

I noticed that on starting WiFi in STA mode on both modules, that micropython displays a message...
'phy_init: failed to load RF calibration data (0x1102), falling back to full calibration' but I would guess that is to be expected given that I did a full flash erase prior to flashing python, thus erasing any WiFi calibration data (stored in PHY NVS?). Note that this message is not displayed on subsequent STA connections which is normal given that WiFi calibration only need to be carried out once.

Following on from proving that both modules are capable of a full WiFi TX signal, I next flashed the master build of CS (on both modules) and observe that once again TX levels are **very** poor.

I am thinking that maybe Wifi calibration data is not being initialised fully by CS given the fact I do a full flash erase before flashing CS? 

I am probably wrong given that no one else has reported any issues with CS on these ESP32 modules, and that I do not see an issue with the NodeMCU part? Why the NodeMCU module does not exhibit the power issue, I do not have any ideas given that all three of my modules are based on the same ESP-WROOM-32 modules. But I have now run out of ideas on how to get my two modules to work reliably with CS. 

One last piece of information - all testing has been carried out with just plain ESP modules, ie no motor driver shield connected, ruling out any hardware interactions with the shield. 

Any pointers as to where to go next would be much appreciated.

atanisoft, 2020-10-22:
micropython uses IDF v3.x last I checked and that may explain some of the differences but not all.

Can you capture the console using `CONFIG_WIFI_DEBUG_OUTPUT=y` in sdkconfig?

Also can you try commenting out these two lines:

- https://github.com/atanisoft/ESP32CommandStation/blob/master/components/OpenMRNLite/src/freertos_drivers/esp32/Esp32WiFiManager.cpp#L1585

- https://github.com/atanisoft/ESP32CommandStation/blob/master/components/OpenMRNLite/src/freertos_drivers/esp32/Esp32WiFiManager.cpp#L1194

I'd suggest erase_flash before testing with these two lines commented.

roadsnail, 2020-10-22:
Yes, the micropython uses IDF v3.something. The reason I flashed it was to check that the two boards are still capable of higher tx power.

Anyhow commented out the lines indicated and rebuilt, flash-erased and flashed... here is the capture...
The tx power level is still down at -80

ESP32 Command Station v1.5.0 (v1.5.0-beta1-11-g1cc1b01-dirty) starting up...
Compiled on Oct 22 2020 16:34:13 using IDF v4.2-beta1-86-gdddcc2ede
Running from: ota_0
ESP32 Command Station uses the OpenMRN library
Copyright (c) 2019-2020, OpenMRN
All rights reserved.
[NVS] Initializing NVS
[TimeZone] UTC0
[ADC] Configure 12-bit ADC resolution
E (783) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (784) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x108 (ESP_ERR_INVALID_RESPONSE) at 0x40087b63
0x40087b63: _esp_error_check_failed_without_abort at C:/Users/chris/Desktop/esp-idf-2/components/esp_common/src/esp_err.c:40

file: "../components/Configuration/FileSystemManager.cpp" line 149
func: FileSystemManager::FileSystemManager()
expression: esp_vfs_fat_sdspi_mount(VFS_MOUNT, &sd_host, &device_cfg, &sd_cfg, &sd_)
[FS] SD Card not present or mounting failed, using SPIFFS
[FS] SPIFFS usage: 17.65/757.66 KiB
[FS] SPIFFS will be used for persistent storage.
[FS] Dumping content of filesystem: /cfg
[FS] /cfg/LCC/cdi.xml (8184 bytes) mtime: Thu Jan  1 00:00:08 1970
[FS] /cfg/LCC/train.xml (5406 bytes) mtime: Thu Jan  1 00:00:09 1970
[FS] /cfg/LCC/tmptrain.xml (2593 bytes) mtime: Thu Jan  1 00:00:09 1970
[FS] /cfg/LCC/config (485 bytes) mtime: Wed Dec 31 23:59:59 1969
[LCC] Initializing Stack (node-id: 5020103f000)
[WiFi] Starting WiFiManager
[Turnout] Initializing DCC Turnout database
[FS] /cfg/ESP32CS/turnouts.json does not exist, returning blank json object
[Turnout] Loaded 0 DCC turnout(s)
[Config] Enabling GPIO Outputs
[Output] Initializing outputs
[FS] /cfg/ESP32CS/outputs.json does not exist, returning blank json object
[Output] Loaded 0 outputs
[Config] Enabling GPIO Inputs
[Sensors] Initializing sensors
[FS] /cfg/ESP32CS/sensors.json does not exist, returning blank json object
[Sensors] Loaded 0 sensors
[Track] Registering /dev/track VFS interface
[OPS] Configuring H-Bridge (L298 2000 mA max) using ADC 1:0
[OPS] Short limit 3686/4096 (1799.80 mA), events (on: 0, off: 0)
[OPS] Shutdown limit 4080/4096 (1992.19 mA), events (on: 0, off: 0)
[PROG] Configuring H-Bridge (L298 2000 mA max) using ADC 1:3
[PROG] Short limit 512/4096 (250.00 mA), events (on: 0, off: 0)
[PROG] Shutdown limit 1024/4096 (500.00 mA), events (on: 0, off: 0)
[PROG] Prog ACK: 122/4096 ( 59.57 mA)
[OPS] DCC config: zero:96uS, one:58uS, preamble-bits:11, wave:high,low
[OPS] signal pin: 19, RMT(ch:0,mem:69[2],clk-div:80,clk-src:APB)
[OPS] Starting signal generator
[PROG] DCC config: zero:96uS, one:58uS, preamble-bits:22, wave:high,low
[PROG] signal pin: 18, RMT(ch:3,mem:80[2],clk-div:80,clk-src:APB)
[PROG] Starting signal generator
[Track] Registering LCC EventConsumer for Track Power (On:10000000000fffe, Off:10000000000ffff)
[eStop] Registering emergency stop handler (On: 10000000000fffd, Off:10000000000fffc)
[Track] Connecting track interface (track:OPS, fd:0)
[Track] Connecting track interface (track:PROG, fd:3)
[CDI] Checking /cfg/LCC/cdi.xml...
[CDI] File /cfg/LCC/cdi.xml appears up-to-date (len 8184 vs 8184)
[CDI] Registering CDI with stack...
[LCC] Config file opened using fd:6
[CDI] Checking /cfg/LCC/train.xml...
[CDI] File /cfg/LCC/train.xml appears up-to-date (len 5406 vs 5406)
[CDI] Checking /cfg/LCC/tmptrain.xml...
[CDI] File /cfg/LCC/tmptrain.xml appears up-to-date (len 2593 vs 2593)
[TrainDB] Initializing...
[... 151 lines cut ...]
I (3219) wifi:connected with msshnot_2, aid = 11, channel 6, BW20, bssid = a2:2a:a8:84:f0:14
I (3219) wifi:security: WPA2-PSK, phy: bgn, rssi: -42
D (3221) wifi:remove all except a2:2a:a8:84:f0:14 from rc list
D (3226) wifi:clear blacklist
D (3229) wifi:filter: set rx policy=7
I (3232) wifi:pm start, type: 1

D (3236) wifi:Send sta connected event
D (3239) wifi:connect status 1 -> 5
D (3242) wifi:obss scan is disabled
D (3245) wifi:start obss scan: obss scan is stopped
I (3250) wifi:AP's beacon interval = 102400 us, DTIM period = 3
[WiFi] Connected to SSID: msshnot_2
[IPv4] [2/36] Waiting for IP address assignment.
[WiFi] IP address is 192.168.20.11, starting hub (if enabled) and uplink
[mDNS] Initializing mDNS system
[mDNS] Setting mDNS hostname to "esp32cs_5020103f000"
I (3758) wifi:Set ps type: 0

[httpd] Starting HTTP listener on port 80
[JMRI] Starting JMRI listener
Listening on port 80, fd 48
Listening on port 2560, fd 49
[mDNS] Advertising _http._tcp:80.
[mDNS] Advertising _esp32cs._tcp:2560.

atanisoft, 2020-10-22:
@roadsnail can you connect to the web interface and on the config page for wifi click on the "Connect to network" button which should perform a WiFi scan. I'm wondering if the RSSI that is reported from that is different than what your AP is showing.

roadsnail, 2020-10-22:
The scan reports the RSSI being -42 being the signal strength of my AP as seen by the ESP32
The signal strength I mentioned above,  (ie -80 ish) is being reported by my AP... ie it's the signal strength of my ESP32 tx

The issue I am still seeing is that the esp32 (espduino) type boards are transmitting on very low power. 

By the way, I would suspect that my previous logged issue (laggy loco commands from JMRI) is probably due to the very low tx level of my CS. 

Thanks for your help so far. I think I will uncomment the two lines commented out earlier, rebuild and continue experimenting here.

atanisoft, 2020-10-22:
> By the way, I would suspect that my previous logged issue (laggy loco commands from JMRI) is probably due to the very low tx level of my CS.

That's possible, it is also possible there is a bug somewhere in the code that is causing a delay.

> Thanks for your help so far. I think I will uncomment the two lines commented out earlier, rebuild and continue experimenting here.

No problem, I'll continue experimentation as well on my side.

TrainzLuvr, 2020-10-22:
@roadsnail for the sake of argument, connect to the **/config** URL of your CS, and look up the _tx_power_ value?

Maybe the value is set low ... and I recall mapping of those values to real dBms was weird, to say the least.

roadsnail, 2020-10-22:
@TrainzLuvr, thanks for the input. I can confirm that tx power in /config reports 20dBm (max value). 

Strangely enough, if I then change the TX power level to 2 and SAVE. Once the CS has rebooted, my AP reports an even lower power level from the CS. So low, that I'm amazed the AP can even communicate with the ESP32! 

I am now suspecting some sort of ESP32 WiFi TX power calibration problem given that flashing micropython on the same hardware resets the maximum tx power level.

TrainzLuvr, 2020-10-22:
@roadsnail err...those values are not in dBms, that was my point.

Espressif chose some weird way of mapping values to actual dBm so that the maximum output power of 20dBm would be a value of 78, check this commit https://github.com/atanisoft/ESP32CommandStation/commit/89f0f177ea2f4401f052f3b836bb62771f6b6bbd

atanisoft, 2020-10-22:
I'm not seeing any specific API in IDF v4.0+ that would explain this degraded RSSI as observed on the AP side, or any reason why the micropython library is able to see better numbers.

roadsnail, 2020-10-23:
**@TrainzLuvr** , yes, thanks again for your input. I got your original point. The 20dBm I referred to was the value displayed in the CS configuration WiFi tx power where I see 20dBm, with a drop-down menu allowing between 2 and 20 tx power to be selected. I get that 2-20 represents min-maximum power values for the wifi tx and not true dBm. Maybe the use of dBm should be dropped in the GUI? Just a thought...  

I presume the 2-20 scale relates to the following (see table below) taken from Espressif source, but whatever, my CS web gui reports that tx power is set to max (20). Whereas my AP (ubiquiti AP AC lite) reports receiving a signal strength  (from ESP32) down around -80dBm (that's proper dBm).

typedef enum {
    WIFI_POWER_19_5dBm = 78,// 19.5dBm
    WIFI_POWER_19dBm = 76,// 19dBm
    WIFI_POWER_18_5dBm = 74,// 18.5dBm
    WIFI_POWER_17dBm = 68,// 17dBm
    WIFI_POWER_15dBm = 60,// 15dBm
    WIFI_POWER_13dBm = 52,// 13dBm
    WIFI_POWER_11dBm = 44,// 11dBm
    WIFI_POWER_8_5dBm = 34,// 8.5dBm
    WIFI_POWER_7dBm = 28,// 7dBm
    WIFI_POWER_5dBm = 20,// 5dBm
    WIFI_POWER_2dBm = 8,// 2dBm
    WIFI_POWER_MINUS_1dBm = -4// -1dBm

**@atanisoft** , once again I appreciate your delving into the 4.0 API docs. I am also trying to get my head around espressif's docs relating to Wifi setup. I am also aware that no one else appears to be having issues with the espduino type boards, so maybe 'park' this issue and see if anyone else reports a similar problem.

One last thing. Am I using a correct ESP-IDF? From the log, my last build shows ...
**Compiled on Oct 22 2020 16:34:13 using IDF v4.2-beta1-86-gdddcc2ede**

Thanks again for your input.

buelowp, 2020-10-23:
I think I can give this a shot. I've got the Wemos Espduino variant. If you think that is close enough? Also, which IDF branch should I build against?

atanisoft, 2020-10-23:
@buelowp IDF v4.0.1 or v4.2 branch should be safe for testing.

roadsnail, 2020-10-23:
@buelowp That would be great if you could give your module a go. This site --  https://www.cnx-software.com/2017/09/04/espduino-32-wemos-d1-r32-esp32-boards-support-some-arduino-uno-shields/ shows pictures of the board type I am using. One of my boards looks exactly like the Wemos versions shown. The other looks identical apart from 'ESPDUINO-32 HW-729' silkscreened where the other shows Wemos...

I have built against V4.2 . My build environment is W10.

atanisoft, 2020-10-23:
@roadsnail I've got a few of the second one (the one branded as HW-729). Functionally they all appear to be identical with exception of the USB port (B vs Micro-B). This is also the same one that I recommend for users as it is one of the easiest to interface with the motor shields etc.

The only downside I know of with it is the PCB antenna on the esp-wroom-32 module is sitting directly on the PCB. It would be better suited if it was elevated slightly or shifted off the PCB so it is not receiving any interference from a copper pour, though in the ones I have it appears they have a copper pour keep-out area under the antenna.

buelowp, 2020-10-23:
Let me know if you want any other details, but here's the capture I just got. My AP is reporting -69 dBm for this connection.

```
et.RÕ¹ 8 &²:2&§Ój

r.ÂE (PêUI=¹_REªbœœÑé0x1&BA%_FA*Õe1SH_šêQ¥
coË¥Í¥p: â SÐR¥Â5
clkEÉÙé0x0°)}ÉÙ:0ø,dEÉÙé0x00iÍÁßÙé0x00¬}Év:0øbºÁ_drö'ÂÁ0
méÄR±œ­div'ŠHë+é0x3fæâleî'ÓHë+é0x3f¢,låK¢ÊÂÂ
ì+é0x40°âleî§¢
ì+é0x40°¢âleKºj
eÉå0x40Â06ŽSHø

ESP32 Command Station v1.5.0 (v1.5.0-beta1-14-ga373938) starting up...
Compiled on Oct 23 2020 07:20:45 using IDF v4.2-beta1-227-gf0e87c933
Running from: ota_0
ESP32 Command Station uses the OpenMRN library
Copyright (c) 2019-2020, OpenMRN
All rights reserved.
[NVS] Initializing NVS
[TimeZone] UTC-5
[ADC] Configure 12-bit ADC resolution
E (802) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (803) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x108 (ESP_ERR_INVALID_RESPONSE) at 0x40087b77
file: "/home/peter/Projects/ESP32CommandStation/components/Configuration/FileSystemManager.cpp" line 149
func: FileSystemManager::FileSystemManager()
expression: esp_vfs_fat_sdspi_mount(VFS_MOUNT, &sd_host, &device_cfg, &sd_cfg, &sd_)
[FS] SD Card not present or mounting failed, using SPIFFS
[FS] SPIFFS usage: 18.38/757.66 KiB
[FS] SPIFFS will be used for persistent storage.
[FS] Dumping content of filesystem: /cfg
[FS] /cfg/ESP32CS/trains.json (433 bytes) mtime: Thu Aug 20 00:09:41 2020
[FS] /cfg/LCC/config (485 bytes) mtime: Thu Jan  1 04:59:59 1970
[FS] /cfg/LCC/cdi.xml (8184 bytes) mtime: Fri Oct 23 18:13:35 2020
[FS] /cfg/LCC/train.xml (5406 bytes) mtime: Fri Oct 23 18:13:35 2020
[FS] /cfg/LCC/tmptrain.xml (2593 bytes) mtime: Fri Oct 23 18:13:35 2020
[LCC] Initializing Stack (node-id: 5020103f000)
[WiFi] Starting WiFiManager
[SNTP] Polling pool.ntp.org for time updates
[Turnout] Initializing DCC Turnout database
[FS] /cfg/ESP32CS/turnouts.json does not exist, returning blank json object
[Turnout] Loaded 0 DCC turnout(s)
[Config] Enabling GPIO Outputs
[Output] Initializing outputs
[FS] /cfg/ESP32CS/outputs.json does not exist, returning blank json object
[Output] Loaded 0 outputs
[Config] Enabling GPIO Inputs
[Sensors] Initializing sensors
[FS] /cfg/ESP32CS/sensors.json does not exist, returning blank json object
[Sensors] Loaded 0 sensors
[Track] Registering /dev/track VFS interface
[OPS] Configuring H-Bridge (L298 2000 mA max) using ADC 1:0
[OPS] Short limit 3686/4096 (1799.80 mA), events (on: 0, off: 0)
[OPS] Shutdown limit 4080/4096 (1992.19 mA), events (on: 0, off: 0)
[PROG] Configuring H-Bridge (L298 2000 mA max) using ADC 1:3
[PROG] Short limit 512/4096 (250.00 mA), events (on: 0, off: 0)
[PROG] Shutdown limit 1024/4096 (500.00 mA), events (on: 0, off: 0)
[PROG] Prog ACK: 122/4096 ( 59.57 mA)
[OPS] DCC config: zero:96uS, one:58uS, preamble-bits:11, wave:high,low
[OPS] signal pin: 19, RMT(ch:0,mem:69[2],clk-div:80,clk-src:APB)
[OPS] Starting signal generator
[PROG] DCC config: zero:96uS, one:58uS, preamble-bits:22, wave:high,low
[PROG] signal pin: 18, RMT(ch:3,mem:80[2],clk-div:80,clk-src:APB)
[PROG] Starting signal generator
[Track] Registering LCC EventConsumer for Track Power (On:10000000000fffe, Off:10000000000ffff)
[eStop] Registering emergency stop handler (On: 10000000000fffd, Off:10000000000fffc)
[Track] Connecting track interface (track:OPS, fd:0)
[Track] Connecting track interface (track:PROG, fd:3)
[CDI] Checking /cfg/LCC/cdi.xml...
[... 439 lines cut ...]
D (2794) wifi:start obss scan: obss scan is stopped
[WiFi] Connected to SSID: buelows
[IPv4] [2/36] Waiting for IP address assignment.
[WiFi] IP address is 172.24.1.56, starting hub (if enabled) and uplink
[mDNS] Initializing mDNS system
[mDNS] Setting mDNS hostname to "esp32cs_5020103f000"
I (2822) wifi:Set ps type: 0

[httpd] Starting HTTP listener on port 80
[JMRI] Starting JMRI listener
Listening on port 80, fd 48
Listening on port 2560, fd 49
I (2838) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[mDNS] Advertising _http._tcp:80.
[mDNS] Advertising _esp32cs._tcp:2560.
Incoming connection from 172.24.1.11, fd 50.
Incoming connection from 172.24.1.11, fd 51.
[SNTP] Received time update, new localtime: Fri Oct 23 18:15:34 2020

Incoming connection from 172.24.1.11, fd 50.
Incoming connection from 172.24.1.11, fd 51.
[TaskMon] uptime: 00:00:45 freeHeap: 174400, largest free block: 113792, tasks: 16, mainBufferPool: 0.67kB
[TaskMon] uptime: 00:01:30 freeHeap: 174400, largest free block: 113792, tasks: 16, mainBufferPool: 0.67kB
[TaskMon] uptime: 00:02:15 freeHeap: 174400, largest free block: 113792, tasks: 16, mainBufferPool: 0.67kB
```

roadsnail, 2020-10-23:
@atanisoft . Yes I agree regarding the Wroom-32 module antenna sitting on the PCB. I too wondered if there was a lovely ground plane running under the antenna. I even broke up some of the PCB, but the little I removed is just plain board. Also, I have a different module, the NodeMCU variant I mentioned earlier, where the Wroom module antenna also sits over the PCB.  This is the variant which delivers full power with the same build flashed! This is driving me nuts!

@buelowp . Thanks for supplying your log. I'll take a look at it and see if I can spot any glaring differences. You mention that your AP sees a signal level of -69dBm. Is that the sort of level you would expect given the distance of your CS from the AP?

buelowp, 2020-10-23:
Yes, it's oscillating between -70 and -66 dBm right now, and I've got a good connection. We have mesh here, so I can tell which node it's attached to, and I can compare that signal strength to other devices on the same node. It's sitting next to my old iPad I use for work, and the iPad is reporting -72 dBm, so it seems to be correct.

roadsnail, 2020-10-23:
@buelowp . Okay thanks. As mentioned above, I see around -80dBm at my access point. (I only have the one). In comparison,  I have an ESP8266 based sensor sitting about 2 feet away from me, and the signal level is reported at -56dBm.

atanisoft, 2020-10-24:
In testing https://github.com/atanisoft/ESP32CommandStation/pull/90 I noticed the following output (from an esp32 that went through factory reset process):
```
[WiFi] Configuring SoftAP (SSID: esp32cs_5020103f000)
[WiFi] Configuring Station (SSID: TheHomeNet)
[WiFi] Starting WiFi stack
[WiFi] Setting ESP32 hostname to "esp32cs_5020103f000".
[WiFi] MAC Address: 30:ae:a4:c1:54:a8
[DHCP] Starting DHCP Client.
[WiFi] Adjusting maximum WiFi TX power 78 -> 46.
[WiFi] Station started, attempting to connect to SSID: TheHomeNet.
```
So something is not 100% correct in the factory reset process in that the maximum TX power should be 78 by default.

roadsnail, 2020-10-24:
> In testing #90 I noticed the following output (from an esp32 that went through factory reset process):
> 
Just to be clear - Are you referring to flash erase, then reflashing espressiff 'AT' binaries?

atanisoft, 2020-10-24:
> Just to be clear - Are you referring to flash erase, then reflashing espressiff 'AT' binaries?

erase flash followed by a reflash of CS binary. I have not use the AT binary on any esp32 (only esp8266).

roadsnail, 2020-10-24:
Okay, thanks. That's interesting. 

I'm currently re-building my build environment just in case there is an issue there. Have just managed to screw things up, so one step forwards two steps back. Ho hum!