atanisoft/ESP32CommandStation #94: Building CS firmware with OLED display enabled fails

TrainzLuvr, 2020-12-27:
I thought to finally connect an OLED display to I2C port and upon setting up the config and trying to build, I got an error:

```
[883/906] Building CXX object esp-idf/StatusDisplay/CMakeFiles/__idf_StatusDisplay.dir/StatusDisplay.cpp.obj
FAILED: esp-idf/StatusDisplay/CMakeFiles/__idf_StatusDisplay.dir/StatusDisplay.cpp.obj
../components/StatusDisplay/StatusDisplay.cpp: In member function 'StateFlowBase::Action StatusDisplay::init()':
../components/StatusDisplay/StatusDisplay.cpp:362:3: warning: missing initializer for member 'i2c_config_t::scl_io_num' [-Wmissing-field-initializers]
   };
   ^
../components/StatusDisplay/StatusDisplay.cpp:362:3: warning: missing initializer for member 'i2c_config_t::scl_pullup_en' [-Wmissing-field-initializers]
../components/StatusDisplay/StatusDisplay.cpp:362:3: warning: missing initializer for member 'i2c_config_t::<anonymous>' [-Wmissing-field-initializers]
../components/StatusDisplay/StatusDisplay.cpp:362:3: error: designator order for field 'i2c_config_t::scl_io_num' does not match declaration order in 'i2c_config_t'
```

TrainzLuvr, 2020-12-27:
Ok, it's not a big deal...just an order of designators.

Correct order in StatusDisplay.cpp is:

```
    .sda_io_num = (gpio_num_t)CONFIG_DISPLAY_SDA,
    .scl_io_num = (gpio_num_t)CONFIG_DISPLAY_SCL,
    .sda_pullup_en = GPIO_PULLUP_ENABLE,
    .scl_pullup_en = GPIO_PULLUP_ENABLE,
```

atanisoft, 2020-12-27:
Which idf version did you use? I thought that I had put in a fix for the ordering already...

atanisoft, 2020-12-27:
This was fixed in https://github.com/atanisoft/ESP32CommandStation/commit/c2b4a9dd815ec3356112819e98646a3dfa433993.

closed 2020-12-27 by atanisoft

TrainzLuvr, 2020-12-27:
The same version I always use now, release/4.2.

TrainzLuvr, 2020-12-27:
Oh, I am still on uplink2 branch, maybe that's why.

atanisoft, 2020-12-27:
Yeah, switch to master latest and it will have the fix. I'll have the code synced soon for other fixes. Web code needs an overhaul as well for CDI access

TrainzLuvr, 2020-12-27:
Yes, new web server for sure!

The one you have on the other nodes is much nicer, leaner and faster. :)

atanisoft, 2020-12-27:
A new web UI overhaul is planned but is a major undertaking. But having at least the CDI interface brought over will be useful.

TrainzLuvr, 2020-12-27:
What is needed to get this stuff going, maybe I can help?

atanisoft, 2020-12-27:
Convert the throttle html to use cash.js and milligram.css mainly. It's not a huge job on that part but the testing and backend parts that will take more time. The biggest part being how best to reimplement the throttle so it is more efficient and can leverage the built in LCC traction protocol.

TrainzLuvr, 2020-12-27:
To be honest, we are both working on our layouts, and should spend more time on it.

I would make the WebUI and throttle secondary to integrating any fixes and stability improvements made in other nodes back into the CS. Having basic features working solid should be the goal: DCC/LCC, Railcom, Accessories, Traction Protocol, CDI.

Even put the Hub aside, seeing how much time we spent on it and the conclusion being that ESP32 just isn't fit for running it for so many reasons. No amount of tweaking is going to make it 100% and, imho, anything less is not accetable, not for the Hub.

atanisoft, 2020-12-27:
Yup, the rework is going to take time and part of it is necessary for pulling in the fixes as the new CDI elements has caused the build time to jump considerably (even with prebuilding the cdi.xml files).

TrainzLuvr, 2020-12-27:
In my case, I need a working CS now because I'm wiring things up...

atanisoft, 2020-12-27:
Yup, let me see if I can get a minimal set together with the latest external code. A few areas will likely be broken in the web UI but I don't think that is critical for your initial testing (since it needs an overhaul anyway)

TrainzLuvr, 2020-12-27:
I'm not worried about a new WebUI atm, as long as the current controls work to turn OPS on/off and do other things.

atanisoft, 2020-12-27:
If you are using CAN/TWAI then you can send an event from the UWT-100 (or JMRI) to turn track on/off. In the web ui the main piece that will not be workable right away will be the config section.

TrainzLuvr, 2020-12-27:
I think we have an issue open here for just that - the power events are not working #86 .

atanisoft, 2020-12-27:
That is due to the event not being delivered or lost somewhere in transit. There have been a few issues related to that which should be fixed with the updated code.