atanisoft/ESP32CommandStation #103: No basic way of checking if a decoder is reachable

sarusso, 2022-01-10:
Hi and congrats for the project.

I built and flashed the latest version (as of today) on an Esp32 Evaluation board. I then hopefully figured out how to use my H-bridge (a classic L298) and connected a LaisDcc decoder directly to the H-bridge output, and in turn a 12V a LED on the decoder motor output.

Now, when I connect to the CS web interface, after adding a loco on address "3" (LaisDcc default) and try to change the speed (in order to hopefully see the LED lighting up), the Speed cursor always get back the zero as soon as i release it.

I acknowledge that there might be a lot of things gone wrong in along the way, so here is the issue: how do I check if the CS just sees my decoder before trying to solve n problems at one?

Thank you!

TrainzLuvr, 2022-01-10:
Excluding any wiring and communications issues between the ESP, the H-bridge and the decoder, this could be a problem with the WebUI itself.

I have had similar problems to yours in the past where the WebUI would not respond or reset. Although I thought that we have isolated and fixed all those since then.

Ideally, if you had a cab/throttle (even your mobile via WiThrottle protocol) please use it to check your physical setup and wiring.

Then next thing you can check is the browser Console tab for any error messages. Another one to check in the browser is the Network tab, and the web socket connection - see if any messages are being sent.

sarusso, 2022-01-10:
Thanks for the hint of using a WiThrottle client. I managed to get my mobile  connected to the Command Station WiFi server, but I cannot set any loco number even there.

However, I recompiled the source code with the option of powering tracks at power in order to at least check if there was any current and there is not. So I am probably having issues with the physical setup too.

TBH, I had an hard time in understanding a lot of options and settings, and i found discrepancies between the documentation and the "idf.py menuconfig" default values. 

I am not sure how make all of this work, there are just too many moving parts. It is kind of a pity, but I understand that it takes a lot of time to write documentation.

Maybe a first step could be to create a beginner how-to using the most standard possible Esp32 board and H-bridge (which I suspect are the ones I have) with pre-compiled, downloadable firmware and pictures of the wiring. 

Thank you, anyway, and feel free to close this.

atanisoft, 2022-01-11:
Which esp32 board and motor board do you have? I do have an issue tracking creation of a user guide and have a Todo item for 2.0 to help make things a bit easier to configure (via selecting hardware up front and defaulting config based on that). It's going to take some time to get that implemented though

atanisoft, 2022-01-11:
> Now, when I connect to the CS web interface, after adding a loco on address "3" (LaisDcc default) and try to change the speed (in order to hopefully see the LED lighting up), the Speed cursor always get back the zero as soon as i release it.

This usually indicates that the WebUI is not communicating with the backend or failed to allocate the loco instance for that address. Can you check the browser console (in developer tools)?

> how do I check if the CS just sees my decoder before trying to solve n problems at one?

Easier said than done, most CS don't know (or care) if a specific decoder is out on the track. They will blindly send packets to whatever address you give it (provided it is a valid address). When it comes to more advanced decoders that support RailCom though they may respond under certain conditions.

> connected a LaisDcc decoder directly to the H-bridge output

I haven't used one of these brand decoders specifically but I do remember some people reporting odd issues with this brand of decoders with some CS. It is not clear why though. Which one do you have specifically so I can check if I can get one and add it to the test bench.

> I had an hard time in understanding a lot of options and settings

Which items did you find as hard to understand?

> i found discrepancies between the documentation and the "idf.py menuconfig" default values.

Most of the defaults are tailored for the custom PCB but I do have known values that can be used for the Arduino form-factor ESP32 boards with Arduino motor shields.  But which options did you find were out of sync between docs and defaults, I'd like to fix that in the documentation.

> However, I recompiled the source code with the option of powering tracks at power in order to at least check if there was any current and there is not. So I am probably having issues with the physical setup too.

What options did you select for pins and where did you connect them to the hardware (ie: ESP32 pin 19 to L298 EN-A etc...)

atanisoft, 2022-01-11:
> I also tried Firefox which is more verbose:
> 
> ```
> WS-RX: {"res":"info","timestamp":"Jan 11 2022 13:02:21","ota":"ota_0","snip_name":"ESP32CommandStation","snip_hw":"1.0.0 esp32","snip_sw":"v2.0.0-alpha1-12-g04","node_id":"5020103fffe","statusLED":true,"statusLEDBrightness":128,"id":1} 192.168.4.1:603:19
> 
> Uncaught ReferenceError: pad_hex is not defined
>     ws_rx http://192.168.4.1/:610
>     ws_rx http://192.168.4.1/:601
> 192.168.4.1:610:14
> 
> The connection to ws://192.168.4.1/ws was interrupted while the page was loading. 2 192.168.4.1:814:13
> ```

Sometimes Chrome hides details like this unfortunately. The `pad_hex` function is defined in `cdi.js` so it should be available but it looks like I will need to move it up to `index.html` instead.

> I thought that there was a sort of acknowledge that a command was received.

That is only applicable on the programming track where the decoder will send an "ack" pulse (60mA+) to the CS. For mainline (OPS) usage though the only option would be via RailCom or other manufacturer specific protocol.

> I connected pin 17 to ENA on the H-Bridge, pin 16 to IN1, and grounded IN2 (which I am not sure if it is correct).

This will not work unfortunately. You will need to connect IN1 and IN2 to the ESP32 but via a small circuit to create an inverse signal between the two pins: https://atanisoft.github.io/ESP32CommandStation/building-esp32cs-idf.html#dcc-signal-splitting

> I also tried to build the v1.5.0-beta1 version just in case using my buildchain (PR #104) but I could not make it compile due to a `error: 'class openlcb::BitEventConsumer' has no member named 'set_state' power_event->set_state(true);` error.

I haven't seen that specific error but it looks likely that the OpenMRNIDF dependency is out of sync with the other code. It would likely be better to stick with master at this point as the v1.5.0 code is effectively EOL after the major rework in v2.0.

> The bridge type and pin numbers in the latest ([04bf6e7](https://github.com/atanisoft/ESP32CommandStation/commit/04bf6e7779d3401f684b561801b26627bab009ea)) commit are completely different: documentation says the default H-Bridge type is the L298 while it now is DRV8873 for OPS and DRV880x for PROG, thee DCC signal/direction pin should be 19 and 18 for OPS and PROG while now it is set to 16, and OPS track enable/pwm pin should be defaulted to 19 while now is to 17. All according to the documentation.

Ahh, yes. The docs need to be updated as the default is now configured for the custom PCB, but it can be used still for the Arduino form factor boards. A note on the DCC signal pin, there is only one now so the docs will need to be updated to reflect that a jumper will be needed between pins on the Arduino Motor Shield (and possibly others). I'll add this to the 2.0 TODO list to clean up.

sarusso, 2022-01-11:
Thank you. I built the circuit and I am doing some testing.

Can you provide me a web browser name / version on which the current web UI works? Even if trying to fix the bug by moving `cdi.js` up in the `index.html` (which does not solve the bug in Firefox by the way) it seems I can't manage to get it work anywhere..

atanisoft, 2022-01-11:
Can you pull the `doc_updates` branch and see if that helps with the WebUI?

I've also added an auto-config option which defines most of the pins out-of-box for a handful of configuration options. With this option it should be possible to go into menuconfig and select `ESP32 with Arduino Motor Shield (L298)` and the config will adjust automatically to the Uno form factor motor shields. I still need to add some documentation rather than rely on it only being on the gh-pages branch (which is out of date as you noted for the 2.0.x versions)

sarusso, 2022-01-11:
So, I tried with the `doc-updates` branch, and the UI kind of works now. I can finally switch the track power on/off button from the UI, but if I try changing the speed slider it always goes back to zero (see the console output in the screenshot).

However, I suspect that I am still doing something wrong with the wiring:

 - on the signal and inverted signal pins, after the circuit I just built following your link, I have about 1.5V on the direct signal and 0.7V on the inverted, and no output from the H-bridge.
 - if I wire the H-bring PWM (ENA) pin directly to a +5V, then I get about 5V out from the H-bridge, with the same voltage on the direct and inverte signal pins on the H-Bridge (still after the circuit I built)

Any hints..?
Thank you.

[image]

atanisoft, 2022-01-11:
That sounds about right I believe for the signal/inverted. It flips polarity at a high rate so most DC polarity checks will be skewed a bit.

Can you share a pic of how you are wiring the L298? It sounds like you are close but missing a couple connections possibly.

sarusso, 2022-01-11:
Sure.

blue wire = PWM
violet wire = direct
gray wire = inverted
brown wires = ground
red wire = +5V

[image]
[image]

sarusso, 2022-01-14:
Do you have any news on this @atanisoft? I do not mean to have a sort of "customer support", but just to know if you see anything macroscopically wrong. Thank you :)

atanisoft, 2022-01-14:
Sorry been busy with other tasks the last few days. I have a handful of those boards and will need to do some further testing but I suspect it may be a hardware issue (wiring or extra components still needed) more than code issue at this point. I do most of my testing of the L298 using the arduino motor shield which includes an inverter chip rather than needing the transistor.

sarusso, 2022-01-14:
No problem as I said I did not expect anything. By "Arduino motor shield"  do you mean this one: https://www.amazon.it/dp/B0084ZQF3O ?

If so, I might just buy that one, and in this case would you be available to share the wiring?

Thank you,
Stefano.

sarusso, 2022-01-14:
Ok, I start to understand. I missed these pictures, are they linked in the docs? However, I just bought the first one you mentioned, and will give one more try with this one  before giving up. I assume that `H-Bridge signal/direction` pin goes to `DIRA` and that `H-Bridge enable/pwm` pin goes to `PWMA`, correct? Then only +5V and ground connections and it should be ready to go? 

p.s. is the `H-Bridge current sense` pin optional or I misunderstood?

atanisoft, 2022-01-14:
> I missed these pictures, are they linked in the docs?

They used to be linked into the old docs tree but got dropped somewhere along the line. https://github.com/atanisoft/ESP32CommandStation/issues/67 and https://github.com/atanisoft/ESP32CommandStation/issues/42 track that work and I'd like to get that covered in the current docs PR.

atanisoft, 2022-01-14:
> @atanisoft if you recall we had a problem using a transistor to invert the signal and thus switched back to digital inverter.

You are right, there was a missing resistor IIRC. Was that on the source pin from the ESP32 to limit current?

TrainzLuvr, 2022-01-14:
This was to invert the OP-SIG and PROG-SIG, so we switched to 74HC08D.

atanisoft, 2022-01-14:
Yes, on the PCB I'm using 74AHC1G08 which is similar but is a smaller package with only one gate

sarusso, 2022-01-15:
Does the green and red marks here mean that I should iron them to create a bridge? https://raw.githubusercontent.com/atanisoft/ESP32CommandStation/v1.2.3/docs/motor-boards/arduino-shield-back.jpg

atanisoft, 2022-01-15:
> Does the green and red marks here mean that I should iron them to create a bridge?

They will have a bridge by default, you will definitely want to break the connection on the VIN (red) otherwise it will feed the power input from the motor shield to the ESP32 board (if you have the Uno form-factor one) and I don't know off hand what connection it will make.  The brake pins (green) is optional to some extent since they can be used for generating a RailCom cut-out, though without the rest of the detector circuit it doesn't make much difference really.

sarusso, 2022-01-15:
Okay, so I tried withe the new motor shield, it still does not work and I have no idea what to to next.

Steps to reproduce: checkout commit ae366ce, build using the buildchain without menuconfig (thus using default values), and in particular:

 - track signal/direction: pin 19
 - OPS track enable/PWM: pin 25

The wiring between the boards is as follows in the picture (brown = GND, green =  track signal/direction, yellow = OPS track enable/PWM, blue/green pair = to tracks)

I am now using a brand new DCC-enabled Roco  locomotive for testing.

Any Ideas? Thanks.

[image]

atanisoft, 2022-01-15:
> The wiring between the boards is as follows in the picture (brown = GND, green = track signal/direction, yellow = OPS track enable/PWM, blue/green pair = to tracks)

Add the following additional wires:
- 5V
- 3v3
- IOREF (to 3v3)
- AREF (to 3v3)
- A0 to IO36 (SVP) -- current sense ops
- A1 to IO39 (SVN) -- current sense prog
- Move your track connection to the middle pair of outputs (A-, A+)

If you enable the track output in WebUI (or via config) you should see the two LEDs behind the middle screw terminals turn on.

sarusso, 2022-01-15:
Nothing. Leds do not light up. Tried manually using a fixed H-bridge conf, they do.

In my setup the tracks were connected to B+ and B- because I was using DIRB and PWMB, however I tried to move it to PWMA and DIRA and using the A+ and A- as well just in case, still nothing.

Look, I am not sure what your target audience for this software is, but in the current status it is unusable for me.

It would be cool if you could:

1. provide a reference setup with the wiring (i.e. a schematic and/or pictures for a specific version of a board, H bridge and power supply all connected together)
2. a pre-built reference firmware version (I still don't know if I have an hardware or software issue)
3. basic documentation on which connection are required for basic functionalities.

I will maybe try a few more things over the next days, but I am giving up for now.

Thank you anyway.

atanisoft, 2022-01-16:
> In my setup the tracks were connected to B+ and B- because I was using DIRB and PWMB, however I tried to move it to PWMA and DIRA and using the A+ and A- as well just in case, still nothing.

Ok, as long as they are in sync and the wiring is correct the LEDs should illuminate when the track is supposed to be powered.

> 1. provide a reference setup with the wiring (i.e. a schematic and/or pictures for a specific version of a board, H bridge and power supply all connected together)

Yes, this is something that is needed still. There used to be a bit of a guide but it went out of date as the project evolved and got dropped along the way as a result. Now however the project is fairly stable with no major restructuring of code planned and the feature set is mostly defined (a few extras are planned but they won't alter the base functionality set).

> 2. a pre-built reference firmware version (I still don't know if I have an hardware or software issue)

The Actions tab contains pre-built binaries using the default configuration settings for various implementations. Starting with the doc update PR I've added a few extra configuration options (PCB specific builds) and I'll be extending this further to include default configuration settings for: L298, LMD18200 and BTS7960 (AKA IBT_2). That will cover the majority of the DIY targeted h-bridges and the custom PCBs that I've been working on (which @TrainzLuvr has an early revision of).

> 3. basic documentation on which connection are required for basic functionalities.

Agreed, this should be covered in the documentation for the custom / DIY assembly as well as how to use off-the-shelf components (ie: Arduino motor shield/clones with Uno form-factor ESP32).

> I am not sure what your target audience for this software is

The intended audience for the software is the tinkerer and enthusiast. Though, it does need more documentation in a few areas to ensure successful deployments in the latest software versions.

> in the current status it is unusable for me.

I've got hardware on my desk for both L298 boards you have tested and I'll be using them to get some better documentation put together. Once I sort out what has gone wrong from what you have posted I'll send an update to here.

sarusso, 2022-01-16:
I did not notice the builds in the actions tab, that's great. A pointer from the docs to pre-built firmware with default values would indeed lower the entry barrier so much IMO. 

Just one note: I noticed older releases (i.e. v1.5.0-beta1, which I would have liked to give a try just in case) are "expired", not sure if it is a GitHub policy or because of your decision, if it is because of GitHub then linking a reference firmware from the docs which might disappear might not be the best way..

Thank you for your help so far. I would have loved to contribute to the docs or writing a blog post somewhere on how to get all of this up and running from a beginner prospective if could make it work. Maybe after you post the update :)

Looking forward for your update then!

atanisoft, 2022-01-16:
> I noticed older releases (i.e. v1.5.0-beta1, which I would have liked to give a try just in case) are "expired", not sure if it is a GitHub policy or because of your decision

This is a default behavior for GitHub Actions. For official releases though they will be attached to the release itself and should be preserved.

atanisoft, 2022-02-01:
I suspect this is the issue with the OPS track not working as you expect:

```
ESP32 Command Station starting up...
Compiled on Jan 29 2022 19:50:34 using IDF v4.4-96-gbfd3e9c54c
Running from: ota_0
ESP32 Command Station uses the OpenMRN library
Copyright (c) 2019-2021, OpenMRN
All rights reserved.
[SNIP] version:4, manufacturer:atanisoft.github.io, model:ESP32CommandStation, hw-v:1.0.0 esp32, sw-v:v2.0.0-alpha1-40-g9f
[CDI] Size: 17683, Version:0200
[SoC] reset reason:1 - power on reset
[SoC] model:ESP32,rev:1,cores:2,flash:No,WiFi:Yes,BLE:Yes,BT:Yes
[SoC] Heap: 280.86kB / 321.60kB
[SoC] App running from: ota_0
[NVS] Initializing RTC vars
[NVS] Initializing NVS
[NVS] Loading configuration
[NVS] Node ID: 05.02.01.03.ff.fe
[NVS] WiFi Mode: SoftAP (2)
[NVS] Hostname Prefix: esp32cs_
[NVS] SoftAP SSID: esp32csap
[NVS] SoftAP Auth: open (0)
[NVS] SNTP: Off
[NVS] FastClock: Disabled
[NVS] LED brightness: 128
[NVS] Bootloader flag is false, start the stack
[FS] SPIFFS usage: 5.15/463.52 KiB
[FS] SPIFFS will be used for persistent storage.
[FS] Dumping content of filesystem: /fs
[FS] /fs/train.xml (3155 bytes) mtime: Thu Jan  1 00:00:04 1970
[FS] /fs/tmptrain.xml (566 bytes) mtime: Thu Jan  1 00:00:05 1970
[FS] /fs/olcb_config (410 bytes) mtime: Wed Dec 31 23:59:59 1969
[ULP-ADC] Registering ULP wakeup ISR
[ULP-ADC] Loading ULP ADC monitoring binary
[ULP-ADC] OPS Short threshold: 3686/4095 (1799.80 mA), Warning threshold: 2764/4095 (1349.61 mA), Shutdown threshold: 4090/4095 (1997.07 mA)
[ULP-ADC] PROG Ack threshold: 122/4095 ( 59.57 mA), Short threshold: 512/4095 (250.00 mA)
[ULP-ADC] Starting the ULP FSM
[ULP-ADC] OPS short detected!
[ULP-ADC] OPS short detected!
[ULP-ADC] OPS short detected!
[ULP-ADC] OPS short detected!
```

This looks likely to be a bug in the ULP code, I'll do some digging and see what I can come up with.

sarusso, 2022-02-02:
I actually have them connected to the correct pins on the Arduino board, I do not have "short detected" messages.. See console output pasted here: https://github.com/atanisoft/ESP32CommandStation/pull/105#issuecomment-1028469108

sarusso, 2022-02-07:
I would say we can close this, as the original issue has basically no solution.

atanisoft, 2022-02-07:
This issue will close automatically with the merge of the PR

closed 2022-02-08 by atanisoft