atanisoft/ESP32CommandStation #109: Add option to disable current sensing for Arduino motor shield

Aleksandr-ru, 2022-08-30:
I'm using [Wemos D1 R32](https://aliexpress.com/item/32916602867.html?spm=a2g2w.orderdetails.0.0.24034aa62bwpD7&sku_id=65979765424&gatewayAdapt=glo2rus) board with [Arduino L298 motor shield](https://aliexpress.com/item/32832947601.html?spm=a2g2w.orderdetails.0.0.11c04aa6ZV06TD&sku_id=64976951888)
and always getting `[ULP-ADC] OPS short detected` message.

If I comment out this check in `components/ULPADC/UlpAdc.cpp` everything runs fine.
If I disconnect half of the motor shield pins - fine too.

Tried to add jumpers from A0 and A1 to another A* pins according to available configuration options - no use.

atanisoft, 2022-08-30:
If you select `ESP32 Command Station Configuration` -> `ESP32 with Arduino Motor Shield (OPS and PROG)` the CS will be automatically configured for the L298 to work with that ESP32 board. You will only need to add a jumper from A0 ->A2 and A1 -> A3.

When selecting the above configuration option it will ignore all other configuration options related to pin assignments.

Aleksandr-ru, 2022-08-30:
> add a jumper from A0 ->A2 and A1 -> A3.

This is magic, it works now, thank you!

But why, why nowhere no big warning like "to get started with L298 motor shield add jumpers"...

atanisoft, 2022-08-30:
The docs cover jumpers in the supported hardware: 
https://github.com/atanisoft/ESP32CommandStation/blob/master/docs/supported_hardware.md#arduino-motor-shield-required-jumpersv

It should also likely be printed in the configuration section as well.

Aleksandr-ru, 2022-08-31:
Thanks again )