atanisoft/ESP32CommandStation #113: Wifi Settings not saving

Nefkens, 2022-10-22:
Hi,

I flashed the images from this page: https://github.com/atanisoft/ESP32CommandStation/actions/runs/3259829163 (ESP32CommandStation-esp32-v4.4-L298)

When i connect to the AP/ESP and open the settings page, the Wifi configuration settings are not saved. I tried setting the Wifi mode to off, station, AP and station, but after a reboot of the ESP the setting is SoftAP Only again. Before the reboot the settings seems to be saved. When i click refresh the value stays at the new value and in the serial monitor i see the new value,
When i change the name in Node Info, that field is saved and persistent after reboot.

I tried two different ESP32: a Uno variant and a nodemcu form factor. Same issue.

atanisoft, 2022-10-23:
Did you press the "Update Complete" button? If not, it is likely the settings won't be persisted.

atanisoft, 2022-10-23:
I've added a couple tags as this should be improved in the documentation and possibly an automatic persist call made after a short period of inactivity after changing configuration settings.

Nefkens, 2022-10-24:
I did press the "Update Complete". At first i didn't use the "Update Complete" button. After a while fiddling around i figured out i needed the press the button. For the page "Node Info" this works, for the Wifi settings it does not. After that i created this issue.

Beside this issue, i think a better name for the "Update Complete" button would be "Save settings".

atanisoft, 2022-10-24:
Ahh, interesting. Looks like there is a bug in the update flow. Let me take a look and get a fix prepared.

Nefkens, 2022-10-25:
I just uploaded the new firmware and it works! The ESP32 is now connected to my wifi network. Thank you for the quick fix.

One last thing, when I load the settings page all input fields have a lot of spaces behind the text. For example "ESP32CommandStation" is shown as "ESP32CommandStation                                           " (note the spaces). When I need to change a value I first have to remove all the spaces (I think).

Or should I open a new issue for this?

atanisoft, 2022-10-25:
Interesting, I hadn't noticed the extra spaces before but that shouldn't be a problem to strip on the UI side. I've pushed an update that should do just that.

atanisoft, 2022-10-25:
I've merged the PR into master and a new build should be available soon. If there are issues please let me know and we can reuse this issue for the same

closed 2022-10-25 by atanisoft