atanisoft/ESP32CommandStation #116: External control (throttle)

Aleksandr-ru, 2022-11-29:
Is there an option to use external control app or device (like dcc++ protocol or something)?
I've found issue https://github.com/atanisoft/ESP32CommandStation/issues/80 about JMRI and DCC++  but there is no documentation about configuration or features (
And next, is there way to use the command station in Rocrail or similar software?

atanisoft, 2022-11-29:
External throttle control is only supported via OpenLCB/LCC Traction Protocol which is supported by JMRI and TCS Throttles.

WiThrottle support is tracked via https://github.com/atanisoft/ESP32CommandStation/issues/15 but there is no specific timeline for it yet.

Aleksandr-ru, 2022-11-29:
So, maybe you can provide a small how-to guide for beginners to setup JMRI or mobile app like Engine Driver or any?

atanisoft, 2022-11-29:
The steps will depend a bit on how/where JMRI is running. If it is running on Linux (rpi or similar) it is easiest to run a standalone OpenLCB Hub and have both JMRI and the CS connect to it, for Windows / Mac it is a bit more complicated but still doable.

However, the rough steps would be something like:
1. Configure JMRI to have an OpenLCB profile.
2. Configure JMRI OpenLCB connection to be "GridConnect over TCP" and supply the IP address (or hostname) where the OpenLCB hub is running. JMRI does include a built-in OpenLCB hub that may be usable as well.
3. Configure JMRI to use OpenLCB as the default option for throttles/turnouts/etc.
4. Start JMRI's built-in WiThrottle server.
5. Navigate to OpenLCB -> Node List and confirm the CS is showing up, if not it may be necessary to configure the OpenLCB uplink settings on the CS and set it to use manual connection mode rather than automatic which searches for the hub via mDNS.
6. Start Engine Driver or compatible application and point it to the JMRI WiThrottle server (it may find it automatically).

After you request a locomotive via Engine Driver confirm that it shows up in the OpenLCB Node List as the CS should create an OpenLCB node for it upon first use (it will also create a roster entry for it on the CS).

I can't guarantee the above steps are 100% accurate, but they should be close. I will work out the exact steps required and post them in the docs tree soon.

atanisoft, 2022-11-30:
> But complicated, especially for newbies.

Agreed, it would be simpler if there was a dedicated device that could take care of the translation between WiThrottle/DCC++/etc and OpenLCB. There are similar bridge devices available commercially but not for OpenLCB, one example being the DigiTrax WiThrottle device.

> Maybe there is a simpler way like DCC++ or similar protocol which requires less setup and less devices in the middle?

DCC++ (and a few other parts) were dropped from v2.0 due to maintenance concerns and limited resources on the esp32 (mostly TCP sockets which is capped at 16 which includes both listeners and clients). About the only way they may be supported would be with an external dedicated device that can take care of translating to/from OpenLCB to the other protocols and there would be some limitations on what is supported in those use cases.

However, there are some options available now that work with ESP32 CS without needing external devices (though a dedicated Hub outside the CS is strongly recommended). The TCS UWT throttles and a couple mobile apps that are in development use the OpenLCB Traction protocol today.