atanisoft/ESP32CommandStation #117: How to use PROG track?

Aleksandr-ru, 2022-12-01:
I'm using ESPDuino-32 with Arduino Motor Shield.
In configuration 
ESP32 / H-Bridge Selection = "ESP32 with Arduino Motor Shield (OPS and PROG)"
Enable OPS track output = checked
Enable PROG track output = checked

But in the web interface there is nothing about PROG track. I've searched sources of web interface and fount nothing about "prog" (like ops_track_power).
What should I do to program CVs of my loco through the command station?

atanisoft, 2022-12-01:
Right now it is not exposed via the web interface unfortunately. I have it on my [todo](https://github.com/atanisoft/ESP32CommandStation/blob/v2-todo/todo.md) list for 2.0 (master is 2.0 beta-1).

I will try and have this implemented on the v2-todo branch next.

Aleksandr-ru, 2022-12-01:
Super! Looking forward for updates)