atanisoft/ESP32CommandStation #118: LM18200 Connections

crood58, 2023-03-21:
I am currently designing a PCB for the ESP32CommandStation. I am currently planning on using two LM18200 H-Bridges. One for operations and the other for program. The ESP32 module is the TTGO ESP32. What pin does the brake output from the H-Bridge get connected on the ESP32. It is unclear in the docs. Is this just for RailCom? Also, I have the CS pin broken out to Pin 36 (ADC0) for operations and to Pin 39 (ADC0) for programing. Is this correct?

Thank you in advance, 

Chris

crood58, 2023-03-21:
Mike, 

I apologize. I didn't realize until this morning that you had already designed a PCB. Do you happen to have the KiCad or gerber files available?

Thanks, 

Chris

atanisoft, 2023-03-21:
> I am currently designing a PCB for the ESP32CommandStation.

I have a completed and tested SMT design and an untested THT design for the CS but I'm unable to release them due to a variety of reasons. Drop me an email and I'll help you with design / reviews.

> I am currently planning on using two LM18200 H-Bridges. One for operations and the other for program.

You might consider an alternative for the PROG track since it is current limited to ~250mA via code.

> What pin does the brake output from the H-Bridge get connected on the ESP32. It is unclear in the docs. Is this just for RailCom?

The brake pin is only used to generate the RailCom cut out. You can tie the brake pin to GND for PROG (if you use the LMD18200T) and similar for OPS if you are not going to enable RailCom. If you are planning to enable the RailCom cut-out you can use a 10k pull-up (3v3) and connect to any free IO pin below 34 which the CS will set to GND when the RailCom cut-out is *NOT* being generated.

> I have the CS pin broken out to Pin 36 (ADC0) for operations and to Pin 39 (ADC0) for programing. Is this correct?

Either of these will work fine, you may need to add a voltage divider and/or op-amp for the current sense pin.

crood58, 2023-03-21:
> I have a completed and tested SMT design and an untested THT design for the CS but I'm unable to release them due to a variety of reasons. Drop me an email and I'll help you with design / reviews. [...]

Mike - thanks for the feedback. I found your schematic and I like your board and you are using the H-Bridge I wanted for the ops. track. I will drop you in email this evening and we can discuss.

closed 2023-03-21 by crood58