atanisoft/ESP32CommandStation #119: meunconfig error

murarduino, 2023-04-07:
[SDK Configuration Editor]
-- Found Git: D:/Esp-tools/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 

[SDK Configuration Editor]
-- ccache will be used for faster recompilation

[SDK Configuration Editor]
-- The C compiler identification is GNU 8.4.0

[SDK Configuration Editor]
-- The CXX compiler identification is GNU 8.4.0

[SDK Configuration Editor]
-- The ASM compiler identification is GNU

[SDK Configuration Editor]
-- Found assembler: D:/Esp-tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe

[SDK Configuration Editor]
-- Detecting C compiler ABI info

[SDK Configuration Editor]
-- Detecting C compiler ABI info - done

[SDK Configuration Editor]
-- Check for working C compiler: D:/Esp-tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped

[SDK Configuration Editor]
-- Detecting C compile features

[SDK Configuration Editor]
-- Detecting C compile features - done

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info - done

[SDK Configuration Editor]
-- Check for working CXX compiler: D:/Esp-tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped

[SDK Configuration Editor]
-- Detecting CXX compile features

[SDK Configuration Editor]
-- Detecting CXX compile features - done

[SDK Configuration Editor]
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32

[SDK Configuration Editor]
---------------------------ERROR--------------------------

CMake Error at D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:452 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 76, in is_solve_required

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Configuring incomplete, errors occurred!
See also "E:/Documents/Espressif-Projects/ESP32CommandStation/build/CMakeFiles/CMakeOutput.log".

[SDK Configuration Editor]
---------------------------ERROR--------------------------

      if project_requirements.manifest_hash != solution.manifest_hash\
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 34, in serialize
      return value.serialize(serialize_default)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 72, in wrapper
      return func(self, *args, **kwargs)
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 159, in get_commit_id_by_ref
      ref = self.run(['ls-remote', '--exit-code', 'origin', 'HEAD'], cwd=bare_path)[:40]
    File "d:\Esp-tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 183, in run
      raise GitCommandError(

  idf_component_tools.git_client.GitCommandError: 'git ls-remote --exit-code
  origin HEAD' failed with exit code 128

  fatal: unable to access 'https://github.com/atanisoft/HttpServer.git/':
  error:1408F10B:SSL routines:ssl3_get_record:wrong version number

  

  

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:32 (project)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

cmake failed with exit code 1

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
Executing action: confserver
Running cmake in directory e:\documents\espressif-projects\esp32commandstation\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 e:\documents\espressif-projects\esp32commandstation"...

[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

atanisoft, 2023-04-07:
Can't say that I've seen that sort of error before. This looks more like an issue with your machine accessing github to download the dependencies as part of the configuration phase of execution.

Are you using a firewall or proxy to route access to the internet? It looks likely that something is intercepting the requests going to github.com and redirecting / altering the request in-flight causing the SSL validations to fail.

murarduino, 2023-04-10:
> Can't say that I've seen that sort of error before. This looks more like an issue with your machine accessing github to download the dependencies as part of the configuration phase of execution. [...]

I don't have much time to spend on managing ESP-IDF. When I install VsCode+IDF and try to create an IDF project, I can open meunconfig normally.
After I downloaded  ESP32 Command Stations (zip)from Github and unzipped it to a directory, I tried to open menuconfig in VsCode and got the above error.

atanisoft, 2023-04-10:
I haven't used VSCode+IDF plugin for anything more than a code editor, I always use command line tools.

I suspect something on your PC (perhaps windows firewall or an antivirus tool) is intercepting the git call to download the dependencies and that is why you are facing issues. Unfortunately, I'm not a windows user so I can't really provide any guidance on how to fix that sort of issue. You might open an issue over on https://github.com/espressif/esp-idf or https://github.com/espressif/idf-component-manager as it relates to downloading dependencies via their tooling.

murarduino, 2023-04-12:
Today I tried to use git clone and run it in IDF CMD mode. First I wasn't sure if I needed a set-target chip (but I did anyway), then I ran idf.py meunconfig, and the problem was as follows:

e:\Documents\Espressif-Projects\ESP32CommandStation>idf.py set-target ESP32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'e:\documents\espressif-projects\esp32commandstation\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory e:\documents\espressif-projects\esp32commandstation\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 e:\documents\espressif-projects\esp32commandstation"...
-- Found Git: D:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:452 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 34, in serialize
      return value.serialize(serialize_default)
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "d:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
[... cut ...]
  fatal: unable to access 'https://github.com/atanisoft/HttpServer.git/':
  error:1408F10B:SSL routines:ssl3_get_record:wrong version number

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:32 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Documents/Espressif-Projects/ESP32CommandStation/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

e:\Documents\Espressif-Projects\ESP32CommandStation>

atanisoft, 2023-04-12:
> idf_component_tools.git_client.GitCommandError: 'git ls-remote --exit-code origin HEAD' failed with exit code 128
> 
> fatal: unable to access 'https://github.com/atanisoft/HttpServer.git/': HTTP/2 stream 1 was not closed cleanly before end of the underlying stream

About the only thing I can come up with is something on your machine is intercepting the git commandline tool and preventing it from working correctly. 

Something you can try:
1. cd "E:\Documents\Espressif-Projects\ESP32CommandStation\components"
2. git clone https://github.com/atanisoft/HttpServer.git --depth 1
3. git clone https://github.com/atanisoft/OpenMRNIDF.git --depth 1
4. edit "E:\Documents\Espressif-Projects\ESP32CommandStation\main\CMakeLists.txt" to add "OpenMRNIDF" and "HttpServer" to the CUSTOM_DEPS list (put each entry on it's own line without the quotes)
5. delete "E:\Documents\Espressif-Projects\ESP32CommandStation\managed_components" directory tree
6. delete "E:\Documents\Espressif-Projects\ESP32CommandStation\depenencies.lock"
7. delete "E:\Documents\Espressif-Projects\ESP32CommandStation\build" directory tree
8. delete "E:\Documents\Espressif-Projects\ESP32CommandStation\main\idf_component.yml"

And then try building, you do not need to call set-target as it will default to esp32 (currently the only supported platform for this project).

murarduino, 2023-04-13:
> About the only thing I can come up with is something on your machine is intercepting the git commandline tool and preventing it from working correctly.
>
> Something you can try:
>
> 1. cd [...]

Step 5, but I can't find the managed_components directory and the dependencies.lock file in the entire ESP32CommandStation directory. Do I need to continue?

murarduino, 2023-04-13:
Skip steps 5 and 6 (can't find relevant directories and files) and rebuild.

E:\Documents\Espressif-Projects\ESP32CommandStation>idf.py build
Executing action: all (aliases: build)
Running cmake in directory e:\documents\espressif-projects\esp32commandstation\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 e:\documents\espressif-projects\esp32commandstation"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: D:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
.....Updating lock file at E:/Documents/Espressif-Projects/ESP32CommandStation\dependencies.lock
Processing 2 dependencies:
[1/2] OpenMRNIDF (a91fe3f613cb2e0ef97764d41997fc33f4e09cc5)
[2/2] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: D:/Espressif/frameworks/esp-idf-v4.4.4/components/freemodbus.
-- Project sdkconfig file E:/Documents/Espressif-Projects/ESP32CommandStation/sdkconfig
Loading defaults file E:/Documents/Espressif-Projects/ESP32CommandStation/sdkconfig.defaults...
E:/Documents/Espressif-Projects/ESP32CommandStation/sdkconfig.defaults:36 CONFIG_ULP_COPROC_ENABLED was replaced with CONFIG_ESP32_ULP_COPROC_ENABLED
-- Found Python3: D:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "ESP32CommandStation" version: 8eb0f2f-dirty
-- Adding linker script E:/Documents/Espressif-Projects/ESP32CommandStation/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v4.4.4/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script E:/Documents/Espressif-Projects/ESP32CommandStation/build/esp-idf/ULPADC/ulp_adc_ops/ulp_adc_ops.ld
-- Automatic configuration of OPS and PROG using L298
-- Components: AccessoryDecoderDB Config DCC HttpServer NeoPixelBus NvsManager OpenMRNExtensions OpenMRNIDF StatusDisplay StatusLED TrainDatabase TrainManager ULPADC Utils app_trace app_update bootloader bootloader_support cbor console cxx driver efuse esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hw_support esp_ipc esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py expat fatfs freertos hal heap ieee802154 jsmn json libsodium log lwip main mbedtls mdns newlib nvs_flash openssl openthread partition_table pthread sdmmc soc spi_flash spiffs tcpip_adapter ulp usb vfs wear_levelling wpa_supplicant xtensa
-- Component paths: E:/Documents/Espressif-Projects/ESP32CommandStation/components/AccessoryDecoderDB E:/Documents/Espressif-Projects/ESP32CommandStation/components/Config E:/Documents/Espressif-Projects/ESP32CommandStation/components/DCC E:/Documents/Espressif-Projects/ESP32CommandStation/components/HttpServer E:/Documents/Espressif-Projects/ESP32CommandStation/components/NeoPixelBus E:/Documents/Espressif-Projects/ESP32CommandStation/components/NvsManager E:/Documents/Espressif-Projects/ESP32CommandStation/components/OpenMRNExtensions E:/Documents/Espressif-Projects/ESP32CommandStation/managed_components/OpenMRNIDF E:/Documents/Espressif-Projects/ESP32CommandStation/components/StatusDisplay E:/Documents/Espressif-Projects/ESP32CommandStation/components/StatusLED E:/Documents/Espressif-Projects/ESP32CommandStation/components/TrainDatabase E:/Documents/Espressif-Projects/ESP32CommandStation/components/TrainManager E:/Documents/Espressif-Projects/ESP32CommandStation/components/ULPADC E:/Documents/Espressif-Projects/ESP32CommandS [... cut ...] Espressif/frameworks/esp-idf-v4.4.4/components/wear_levelling D:/Espressif/frameworks/esp-idf-v4.4.4/components/wpa_supplicant D:/Espressif/frameworks/esp-idf-v4.4.4/components/xtensa
-- SNIP: ESP32CommandStation/atanisoft.github.io/8eb0f2f-dirty/1.0.0
CMake Error at CMakeLists.txt:98 (message):
  Unable to find 'gzip' program

-- Configuring incomplete, errors occurred!
See also "E:/Documents/Espressif-Projects/ESP32CommandStation/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

atanisoft, 2023-04-13:
> Step 5, but I can't find the managed_components directory and the dependencies.lock file in the entire ESP32CommandStation directory. Do I need to continue?

That should be fine, they would only exist if the IDF dependency download worked correctly. Since they did not exist it's fine to skip.

> -- SNIP: ESP32CommandStation/atanisoft.github.io/8eb0f2f-dirty/1.0.0
> CMake Error at CMakeLists.txt:98 (message):
> Unable to find 'gzip' program

This should be fixed on the latest master code, grab the top level [CMakeLists.txt](https://github.com/atanisoft/ESP32CommandStation/blob/master/CMakeLists.txt) and replace your local copy with it. The original CMakeLists.txt used an external copy of gzip to compress the web content and this new one moves to using a CMake built-in compression target.

> .....Updating lock file at E:/Documents/Espressif-Projects/ESP32CommandStation\dependencies.lock
> Processing 2 dependencies:
> [1/2] OpenMRNIDF (a91fe3f613cb2e0ef97764d41997fc33f4e09cc5)
> [2/2] idf (4.4.4)

This is interesting, it seems that it did find OpenMRNIDF via IDF dependency checks but failed for HttpServer. Did you clone both components to the components tree?

murarduino, 2023-04-14:
> That should be fine, they would only exist if the IDF dependency download worked correctly. Since they did not exist it's fine to skip.
>
>
> This should be fixed on the latest master code, grab the top [...]

I've cloned HttpServer, but I don't know why it's not called

E:\Documents\Espressif-Projects\ESP32CommandStation\components\HttpServer>dir
 驱动器 E 中的卷是 docs
 卷的序列号是 38FB-CDDE

 E:\Documents\Espressif-Projects\ESP32CommandStation\components\HttpServer 的目录

2023/04/13  16:41    <DIR>          .
2023/04/13  16:41    <DIR>          ..
2023/04/13  16:41               949 CMakeLists.txt
2023/04/13  16:41               230 idf_component.yml
2023/04/13  16:41    <DIR>          include
2023/04/13  16:41             2,859 Kconfig
2023/04/13  16:41             1,347 LICENSE
2023/04/13  16:41               270 README.md
2023/04/13  16:41    <DIR>          src

CMakeLists.txt:
.......
# starting in IDF v5 esp_ipc component has been merged into esp_system
if (IDF_VERSION_MAJOR LESS 5)
    list(APPEND IDF_DEPS esp_ipc)
endif()

set(CUSTOM_DEPS
    AccessoryDecoderDB
    Config
    DCC
    NeoPixelBus
    NvsManager
    OpenMRNExtensions
    StatusDisplay
    StatusLED
    TrainDatabase
    TrainManager
    ULPADC
    Utils
    OpenMRNIDF
    HttpServer
)
.........

murarduino, 2023-04-14:
I just did it again, and it finally compiled correctly. But I still have some questions to ask:
1. Is OpenMRNIDF and HttpServer components required?
2. How can I define a multi-class motor module, e.g. PROG using L298 and MAIN using BTS7960B? Also, can the motor control pins be customized?

atanisoft, 2023-04-14:
> * Is OpenMRNIDF and HttpServer components required?

Yes, OpenMRNIDF is providing OpenLCB/LCC functionality as well as portions of the DCC management. HttpServer is used for the built-in management interface. They can't really be excluded.

> * How can I define a multi-class motor module, e.g. PROG using L298 and MAIN using BTS7960B? Also, can the motor control pins be customized?

In menu config, in the H-Bridge selection choose the [custom](https://github.com/atanisoft/ESP32CommandStation/blob/master/components/Config/Kconfig.projbuild#L55) option. Set the OPS H-Bridge type and PROG H-Bridge type as you are wanting and it should work. However, I've not had much luck with the BTS7960B boards that I purchased and use L298/LMD18200 for my usages, if I need higher current limits I use an external booster.

murarduino, 2023-04-14:
> Yes, OpenMRNIDF is providing OpenLCB/LCC functionality as well as portions of the DCC management. HttpServer is used for the built-in management interface. They can't really be excluded.
>
>
> In menu [...]

Thanks to your guidance, the ESP32CS has been built correctly in three different environments.
I am learning the configuration in menuconfig. Now I have a Uno motor board for L298, but no ESP32 board for Uno (only one ESP32-DevKitC), so I want to ask how to connect them?

atanisoft, 2023-04-14:
> so I want to ask how to connect them?

Pin map is [here](https://github.com/atanisoft/ESP32CommandStation/blob/master/components/Config/include/pinmap-l298.hxx#L67-L83), you will need to connect them to the corresponding pins on the motor shield and connect 5v/3v3/GND. DCC Signal should go to the DIR pins, DCC Enable to PWM, A0 -> 35, A1 -> 34.

murarduino, 2023-04-14:
so 
single_pin: GPIO19 -> motor board 12 and 13
OPS_pin: GPIO 25 ->  motor board 3 
PROG_pin:GOIO 23 -> motor board 11
right?

atanisoft, 2023-04-14:
you will also need 3v3 (ioref), 5v and GND connected.

atanisoft, 2023-04-14:
possibly a few others, I haven't hooked up a devkitc board to a shield in quite some time

murarduino, 2023-04-16:
Does the dependency not contain an HttpServer pair?
When menuconfig or build, Processing dependencies only include OpenMRNIDF and idf, not HttpServer.

dependencies.lock:
dependencies:
  HttpServer:
    component_hash: 07f24e13f58488ebabd54b084b77b6f8825075ed70541656481e9956b449102e
    source:
      git: https://github.com/atanisoft/HttpServer.git
      path: .
      type: git
    version: f06c0f296bb0a3e54ca00db6b46a1b904fd849fb
  OpenMRNIDF:
    component_hash: f03af9a08568e0c48aaecdae630da799a5bd14ed01080af10afa3f31d9b285b7
    source:
      git: https://github.com/atanisoft/OpenMRNIDF.git
      path: .
      type: git
    version: a91fe3f613cb2e0ef97764d41997fc33f4e09cc5
  idf:
    component_hash: null
    source:
      type: idf
    version: 4.4.4
manifest_hash: b8334d372cbbd6388ea2af7488c38e0ef70475db7de803544c9e73f6a247fc40
target: esp32
version: 1.0.0

D:\esp>git  clone https://github.com/atanisoft/ESP32CommandStation.git
Cloning into 'ESP32CommandStation'...
remote: Enumerating objects: 6343, done.
remote: Counting objects: 100% (885/885), done.
remote: Compressing objects: 100% (433/433), done.
remote: Total 6343 (delta 555), reused 532 (delta 446), pack-reused 5458
Receiving objects: 100% (6343/6343), 47.11 MiB | 4.51 MiB/s, done.
Resolving deltas: 100% (4224/4224), done.

D:\esp>cd  ESP32CommandStation

D:\esp\ESP32CommandStation>cd  components

D:\esp\ESP32CommandStation\components>git clone https://github.com/atanisoft/HttpServer.git --depth 1
Cloning into 'HttpServer'...
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 19 (delta 2), reused 3 (delta 0), pack-reused 0Receiving objects:  84% (16/19)
Receiving objects: 100% (19/19), 44.51 KiB | 403.00 KiB/s, done.
Resolving deltas: 100% (2/2), done.

D:\esp\ESP32CommandStation\components>git clone https://github.com/atanisoft/OpenMRNIDF.git --depth 1
Cloning into 'OpenMRNIDF'...
remote: Enumerating objects: 393, done.
remote: Counting objects: 100% (393/393), done.
remote: Compressing objects: 100% (358/358), done.
remote: Total 393 (delta 102), reused 163 (delta 32), pack-reused 0
Receiving objects: 100% (393/393), 876.88 KiB | 870.00 KiB/s, done.
Resolving deltas: 100% (102/102), done.

D:\esp\ESP32CommandStation\components>cd ..

D:\esp\ESP32CommandStation>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory d:\esp\esp32commandstation\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 d:\esp\esp32commandstation"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: D:/esp/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
[... 33 lines cut ...]
-- Adding linker script D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script D:/esp/ESP32CommandStation/build/esp-idf/ULPADC/ulp_adc_ops/ulp_adc_ops.ld
-- Automatic configuration of OPS and PROG using L298
-- Components: AccessoryDecoderDB Config DCC HttpServer NeoPixelBus NvsManager OpenMRNExtensions OpenMRNIDF StatusDisplay StatusLED TrainDatabase TrainManager ULPADC Utils app_trace app_update bootloader bootloader_support cbor console cxx driver efuse esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hw_support esp_ipc esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py expat fatfs freertos hal heap ieee802154 jsmn json libsodium log lwip main mbedtls mdns newlib nvs_flash openssl openthread partition_table pthread sdmmc soc spi_flash spiffs tcpip_adapter ulp usb vfs wear_levelling wpa_supplicant xtensa
-- Component paths: D:/esp/ESP32CommandStation/components/AccessoryDecoderDB D:/esp/ESP32CommandStation/components/Config D:/esp/ESP32CommandStation/components/DCC D:/esp/ESP32CommandStation/components/HttpServer D:/esp/ESP32CommandStation/components/NeoPixelBus D:/esp/ESP32CommandStation/components/NvsManager D:/esp/ESP32CommandStation/components/OpenMRNExtensions D:/esp/ESP32CommandStation/managed_components/OpenMRNIDF D:/esp/ESP32CommandStation/components/StatusDisplay D:/esp/ESP32CommandStation/components/StatusLED D:/esp/ESP32CommandStation/components/TrainDatabase D:/esp/ESP32CommandStation/components/TrainManager D:/esp/ESP32CommandStation/components/ULPADC D:/esp/ESP32CommandStation/components/Utils D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/app_trace D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/app_update D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/bootloader D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/bootloader_support D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/cbor D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/console D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/cxx D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/driver D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/efuse D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/esp32 D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/esp_adc_cal D:/esp/Espressi [... cut ...]
D:/esp/ESP32CommandStation/sdkconfig.defaults:36 CONFIG_ULP_COPROC_ENABLED was replaced with CONFIG_ESP32_ULP_COPROC_ENABLED
D:/esp/Espressif/frameworks/esp-idf-v4.4.4/Kconfig:14: warning: IDF_ENV_FPGA has 'option env="IDF_ENV_FPGA"', but the environment variable IDF_ENV_FPGA is not set
Loaded configuration 'D:/esp/ESP32CommandStation/sdkconfig'
Configuration saved to 'D:/esp/ESP32CommandStation/sdkconfig'
Loading defaults file D:/esp/ESP32CommandStation/sdkconfig.defaults...
D:/esp/ESP32CommandStation/sdkconfig.defaults:36 CONFIG_ULP_COPROC_ENABLED was replaced with CONFIG_ESP32_ULP_COPROC_ENABLED

atanisoft, 2023-04-16:
Normally HttpServer is automatically included as it exists in idf_component.yml as seen in [this](https://github.com/atanisoft/ESP32CommandStation/actions/runs/4708589766/jobs/8351113859#step:6:91) line from the recent build test. However, due to your reported error in resolving the dependencies via the normal method and switching to cloning HttpServer into the components tree and removing the reference from idf_component.yml it will not show as a dependency alongside idf and OpenMRNIDF (which it is successfully cloning as part of the dependency download).

It is not clear why HttpServer is failing to download as part of the dependencies automatically. I suspect something running on your machine might be intercepting it as malware or similar which is a definite false positive.

murarduino, 2023-04-17:
After a few more rounds of testing, I finally understood the problem with the HttpServer and OpenMRNIDF packages.

First, main\idf_component.yml contains information about the dependent component, which will be downloaded from Git when building or menuconfig. The previous problem may be that git was not accessible at the time during this process, so that the two Repositories could not be installed correctly.

Secondly, when installing these two components manually, HttpServer is not reflected in dependences.lock, but it does not seem to affect the use, I can still build or menuconfig normally

Finally, to prevent problems during configuration and to make it easier, I copied the three Repositories to my repository in gitee and added two URLs to git config --global:

git config --global url.https://gitee.com/muralb/OpenMRNIDF.git.insteadOf https://github.com/atanisoft/OpenMRNIDF.git
git config --global url.https://gitee.com/muralb/HttpServer.git.insteadOf https://github.com/atanisoft/HttpServer.git

This will automatically deploy and work (the two lazy packages will display correctly).
Copy your three repositories just want to be able to better use, please understand and hope to get your support. With your permission, I'll expose these three libraries in gitee.

In addition, there is a question for you:Regardless of which H-bridge is selected, the SDA pin of I2C is 23, which conflicts with the PROG_Pin of L298. So, every time I reconfigure menuconfig, I change the SDA_pin of the I2C to 22.

atanisoft, 2023-04-17:
> Regardless of which H-bridge is selected, the SDA pin of I2C is 23, which conflicts with the PROG_Pin of L298.

That is only a default value, it can be adjusted as you have found. For the automatic configuration options it shifts between 22 and 23 depending on which option is picked. 

> With your permission, I'll expose these three libraries in gitee.

Please do not publish these outside of GitHub, they will not be supported or maintained and it will ultimately create confusion for users.

> HttpServer is not reflected in dependences.lock

This is expected as it doesn't have a reference in idf_component.yml with the workaround you have applied. This file (and all files under managed_components) come from the idf_component.yml file references.

> main\idf_component.yml contains information about the dependent component

*EXTERNAL* dependencies, it will not include all dependencies.

> The previous problem may be that git was not accessible at the time during this process

Yes, as your logs indicate it had failures during cloning of HttpServer.git which appear to be due to a firewall or other external influence causing SSL failures.

murarduino, 2023-04-18:
> OK, I'll take your advice and use it myself.
Thanks again for your guidance. I think this Issue can be closed.

closed 2023-04-18 by murarduino