atanisoft/ESP32CommandStation #121: Get started

jbanety, 2023-08-09:
Hi @atanisoft ,
Your project looks very promising and I'll give it a try.
But I'm really new to the world of model railroading and even more so with DCC and the ESP32. can you give me instructions to start?
With this, I'll try to update the Get Started Doc with a PR 😉

atanisoft, 2023-08-09:
I'll gladly accept PRs for the docs, right now this project is mostly in maintenance only mode since it is fairly stable (there are some bugs) and I don't have time to focus exclusively on this project.

The original intent of the getting started guide was going to be picking up from the "I've got the hardware configured and software loaded onto the esp32, what's next" point. I was going to add some notes on connecting JMRI to the CS and extending the functionality provided by both. I was also going to add some documentation about the web interface and how to use various parts of it.

Feel free to send me an email and we can discuss further.

jbanety, 2023-08-09:
Well, sounds good to me.
I early backed the new Udoo Key, a board combining Raspberry Pi RP2040 microcontroller and ESP32-WROVER-E module.
My goal is to set up a light automation of my first track plan with it.